bambulab/BambuStudio #763: Allow 'Price money/kg' field to be editable in system filament presets

AimeeAllen, 2022-12-04:
Bambu Studio comes with many useful system filament presets, both for Bambu Lab specific filaments, and also generic filaments. Whilst I completely understand why almost all fields should be read-only, it would be wonderful if the price money/kg field could be editable and saveable for each user.

Bambu users are from all over the world, and we shop in a handful of different currencies on your various Bambu stores around the world, so having one fixed money/kg does not make sense. I personally really like the price estimates provided to decide if a particular print is economically viable, but I don't want to have to make a copy of every single filament that I use when the rest of the presets are wonderful. Similarly, there are huge variances in the price for some generic products (e.g. Generic PLA in my local currency varies between $17 and $39/kg) and so some people may want to edit this to suit their most commonly purchased filaments.

As a minimum (if we are unable to edit the price field), if would be wonderful if it could show the price for our local region e.g. if a user could select "Show prices for EU store" or "Show prices for AU store" to see more regionally accurate prices for their Bambu filament usage.

Benderrodrigu3z, 2022-12-04:
You can already do this if you save the profile, if you mean change the default one that would set it default price for everyone however you can go into the file system and change the predefined number but it gets overwritten on the next update.

Merlwynd, 2022-12-04:
I support this request. I realize one can save another the profile as another preset (or change it every time you print which is annoying)....However, I'm not sure where the default profile lives...if it lives on my computer (i.e. locally saved when I download Bambu Studio) then this should be possible, but if it lives in cloud and Bambu Studio accesses it from there whenever Studio is in use, then I guess this cannot be done as it would change the default profile for everyone...in which case only viable option is to save default profile as a local user preset with correct price.....

Benderrodrigu3z, 2022-12-05:
> I support this request. I realize one can save another the profile as another preset (or change it every time you print which is annoying)....However, I'm not sure where the default profile [...]

It lives on your computer in a hidden folder, on my its inside the library folder but its the default profile because they want you to have a way to revert back to a known good setting where someone hasn't touched it

Benderrodrigu3z, 2022-12-05:
On Mac its in Home User Folder/Library/Application Support/System/BBL/Filament/Generic PLA.json

Benderrodrigu3z, 2022-12-05:
{
    "filament_cost": [
        "27"
    ],
    "filament_settings_id": [
        "pla Rainbow SUNLU $27"
    ],
    "from": "User",
    "inherits": "Generic PLA",
    "name": "pla Rainbow SUNLU $27",
    "version": "1.3.0.2"
}

chrbratt, 2023-04-09:
> Bambu Studio comes with many useful system filament presets, both for Bambu Lab specific filaments, and also generic filaments. Whilst I completely understand why almost all fields should be [...]

+1 for a feature to be able to handle currency. The easiest thing would be to be able to have a global function for currency and where you can enter your currency rate against the euro, which seems to be the default currency.

ShortyCM, 2024-05-02:
Yeah, I can see the benefit of the default profiles not being editable, but the price should certainly be exempt from this.  I don't necessarily care what the price tag in the store is.  I want to include tax and shipping, my true cost for the filament, not just the price tag.  Saving a user profile is all well and good, but if I don't plan on changing anything else in the default profile it means I then have to remember that I need to load the user profile when I change the filament, since I presume it loads the default profile when it finds the RFID tag after loading the new roll.  It's a minor inconvenience, but an inconvenience none the less.  This should be a user-editable portion of the otherwise read-only default profiles.

Merlwynd, 2024-05-03:
+1 to be able to change the filament price only on default profiles for the reasons given above.