bambulab/BambuStudio #764: Add ability for AMS to detect how much filament is left on a spool, and display this within Bambu Studio

AimeeAllen, 2022-12-04:
This is more of a longterm feature request, as (to my current knowledge) AMS lacks some of the hardware (sensors) required to achieve this.

It would be extremely useful if the AMS system had a way to tell how much filament was left on each spool, and pass this information onto Bambu Studio, Bambu handy and the printer touch screen. In particular, having this information available in Bambu studio would better allow users to select a spool that was not going to run-out mid-print job, or to have an estimate of when it would run out so they could change it over without the print getting too cold.

A nice extension (altenative?) would also be to allow an option to choose what happens when a spool runs out e.g. pause and ask user to change the spool, or to switch to a different numbered spool already in the AMS.

AbeFM, 2022-12-07:
Useless is a strong word. I have this function on the RAISE3D E2 at work. To be honest, as c2h mentioned, it works out pretty infrequently - in the real world, I end up swapping a roll more often than I use one up. Of course, that printer is direct feed, while rolls do tend to hang out in the AMS longer.

The option of going to the next spool when a spool runs out has been the topic of suggestions here, on the forums, on discord and has been asked for directly in a few influencer videos on the YouTubes.... And I've never seen an acknowledgement of it from Bambu. From a layman's perspective, it seems pretty easy to add.

Perhaps a compromise would be to **display "meters used since loading" on the AMS screen, under each roll**. You, as the user, can decide how much is left, but it would be nice to know if 12m or 115m came off so far.

c2h5oh, 2022-12-07:
> The amount of filament on a spool is consistent (and generally an overage not an underage), at least with reputable brands. The change the industry needs is printing empty spool weight on the spools. :-)

Polymaker allows +/-2.7% variance, eSun, Sunlu +/-5% iirc and so on. Most will set machines to "slightly over the number on the label" and not worry about it. Enough variance to make estimating if you have enough near the end of the spool too risky.

I do agree that keeping track of amount used from a spool makes sense - you add a new spool, it gets auto-assigned a new ID that you sharpie on it, usage get tracked. When you re-add it to AMD you provide that ID and it continues from there. Added bonus: you can tie flow calibration results to that ID too.

AbeFM, 2022-12-07:
Perhaps I'm just getting lucky that most of my spools (mostly don't use Polymaker and very little esun) are at-or-above. It's great they give a spec.

The point I was trying to make is that, even without any ID's, tracking, assumptions about weight/density/material or anything, you could display the amount fed since a particular roll went into the machine. Assuming I didn't take some for another machine while it was out would lead to more problems. I'd recommend a simple meters-since-loaded counter unless you have serialized spools/RFID's, even then it's a guess without a scale.

Wookbert, 2023-12-12:
The EEPROM-equipped SmartReels of my 2015 [CEL Robox FDM printer](https://cel-robox.com/) were not only re-writable after e.g. a refill, but also held pretty accurate info on how much filament was left on the spool in meters and grams.