bambulab/BambuStudio #768: Add ability to program AMS from PC

AbeFM, 2022-12-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The touch screen can be a little slow, it would be nice to be able to set a filament type/color from BS

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
be able to set a filament type/color from BS

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Getting up, walking to machine, changing it there

**Additional context**
Add any other context or screenshots about the feature request here.

c2h5oh, 2022-12-04:
You can already do this. Bambu Studio -> Device tab -> right hand side:

[image]

AbeFM, 2022-12-05:
I just came here to add that. Thanks for the catch.

AbeFM, 2022-12-05:
Do'oh! I'd seen that a bunch of times and never clicked on it.

Er, thanks!

closed 2022-12-05 by AbeFM