bambulab/BambuStudio #772: [Feature Request] Suggestions for improving the Seam Painting tool

tokamac, 2022-12-05:
**Seam Painting** is a fantastic feature of BambuStudio, but in its current form is a pain to use. For example: 

- When zooming in a model to select the portion where one wants to paint a seam, switching from the Prepare view to the Painting view resets the view to completely another place. 

- Magnifying the model zooms in from a point near the center of the screen, instead of the point where the cursor is currently located. 

- Conversely, rotating the model occurs around its geometric center, meaning that if you want to revolve around a particular area located at the periphery of a large model, rotating the whole model makes this area of interest escape out of the viewport at light speed and you loose it.

- Accidentally clicking on the model instead of a hollow area around it when panning/zooming instead of painting immediately quits the Painting Seam tool and we unexpectedly land back into the Prepare view again.

→ Consequence from all these UI behaviors: an unnecessary large number of successive trials, panning and zooming in/out steps are required to reach a particular spot.

**Suggestions:**

1. Solve the resetting location issue when switching from normal view (Prepare) to the painting view. The location and perspective should be exactly the same in both panes, as they are when switching from the Prepare view to Preview.

2. Zoom in the model where the cursor is actually is. It would be so much better to aim a particular point in the model! 

3. Make the cursor location the center of rotation when revolving around the model, for the same reason as point (2).

4. Add a button on screen to let the user choose when he really wants to quit the Painting Seam tool, not letting him accidentally return to the Prepare view, loosing all he was trying to achieve.

5. This one is maybe a bit tricky to implement  but would be wonderful: Let the user load a 3D model as a "Seam modifier" that would tell BambuStudio to try to draw the seam along the loaded shape. In this cas the user would not draw the seam in BambuStudio manually in an uncertain way, but very precisely in a CAD software, and would "import the seam path" into BambuStudio.

randyre, 2022-12-05:
Or how about a path tool that lets you draw straight lines connected to each other kind of like the Photoshop line tool.?

tokamac, 2022-12-05:
> Or how about a path tool that lets you draw straight lines connected to each other kind of like the Photoshop line tool.?

Excellent idea, yes.

ResiduaLiu, 2022-12-09:
Nice suggestions. Cannot agree more on 4. Everytime I zoom in the object，I need to find the empty space on screen for rotation and the object is still rotating around its mass center.

eeverman, 2025-02-27:
It would also be nice to be able to do a cross-section view to get to hard to see areas to do seam painting.  I don't think there is currently a way to control seams on areas you can't directly view, other than the basic aligned or back options.