bambulab/BambuStudio #778: More Support Filament options.

randyre, 2022-12-06:
**Is your feature request related to a problem? Please describe.**
Trying to keep filament switching down and use less support-w filament.

**Describe the solution you'd like**
More options for using support-w as interface material.

**A clear and concise description of what you want to happen.**
If we could have either an option to only use support-w filament interface on large flat surfaces, or have an option when painting supports to paint normal supports with the standard interface or support-w filament interface.

ResiduaLiu, 2022-12-09:
So in general what you want is adding material properties to support painting, right?

randyre, 2022-12-10:
@ResiduaLiu I think that's right, What I want is to be able to paint supports with different AMS slots just like colored painting does but decide between the entire support and just the interface. I only use support-w on the actual interface to conserve it as the rest of the support doesn't matter what its printed in if the support-w material is there.  
Lots of times standard overhangs work just fine with standard PLA as support interface, but large flat surfaces, or parts with very small gaps really benefit from Support-w interface, but not having to have the entire model in one or the other would be super beneficial.

randyre, 2023-02-21:
@ResiduaLiu   Any chance this will be looked at in an upcoming release?

SaltWei, 2023-10-08:
Similar to https://github.com/bambulab/BambuStudio/issues/411