bambulab/BambuStudio #779: How much RAM do you have on your machine?

WZ9V, 2022-12-07:
How much RAM do you have on your machine?

_Originally posted by @sigxcpu76 in https://github.com/bambulab/BambuStudio/issues/12#issuecomment-1337970754_
      
16GB Linux Mint 21 Cinnamon

deftdawg, 2022-12-07:
I was able to build in a Virtualbox with 12GB of RAM running nixOS (the same config with only 4GB failed), my guess is you'll need 8GB to 10GB of FREE mem to build...  so if you have anything other than a DE running you might want to close it or reboot to get the free memory cleaned up.

WZ9V, 2022-12-09:
Thanks for the help.  This fixed official but I noticed the softfever version fails at the copy step in the container file.  I'll jump over there for that.

PR #695 (2022-12-09): AppImage Build fixes for Podman / Docker
AppImage Build Fixes:

1. @sigxcpu76's workaround to force GTK Theme to be light, so all text is visible applied to AppRun
2. @SR-G's docker building syntax fixes and instructions to allow building with docker
3. Updates to podman syntax; sudo not required
4. Container rename to bambu-studio-builder to avoid potential confusion
5. Free memory and disk checking in buildLinux.sh to prevent compiles failing with error 2 (not enough memory) or gcc from failing with out of disk space.

Fixes #779, #684, #758, #653
closed 2022-12-09 by WZ9V