bambulab/BambuStudio #780: Issues caused by supports being printed at a different layer height than the model

kvnper, 2022-12-07:
Hi,

I have attached the project file here for your reference.

[BBS support height and speed issue v1.zip](https://github.com/bambulab/BambuStudio/files/10172651/BBS.support.height.and.speed.issue.v1.zip)

Issue description: The printed model has different colour layers every few layers (see pic below).

[image]

Slicer view of speed: 
[image]

I have found a cause of the issue in the slicer. The supports are being printed at a different, higher layer height than the model layer height. This means the support layers stack higher faster than the model, and the slicer will periodically (every few layers) print the model's layers twice in a row in order to catch up to the supports layer. This causes the printer to slow down from 120mm/s to 20mm/s for the second layer.

In the pictures below I will show a side view of layers 17 to 19 where you can observe this occurring (see pic below).

Layer 17:
[image])

Layer 18:
[image]
Layer 18 - Zoomed in:
[image]

Layer 19:
[image]
Layer 19 - Zoomed in:
[image]
Layer 19 - Zoomed in + Line type view:
[image]

Contextual Photos:
[image]
[image]

MusangKingBambulab, 2023-02-13:
Hi, @kvnper I got your problem. We will provide an option for the synchronized layer with support layers. Thanks for your advice.

nmkr, 2024-07-07:
@MusangKingBambulab is the function implemented yet?