bambulab/BambuStudio #781: Support interface not being generated on angled wall

kvnper, 2022-12-07:
Project file attached here:
[BBS support height and speed issue v1.zip](https://github.com/bambulab/BambuStudio/files/10172841/BBS.support.height.and.speed.issue.v1.zip)

**Issue: I have an angled wall, and the support structure does not generate a support interface for it.**

I have tried increasing the support angle and support painting the mesh but with no success.

See pics below.

[image]

[image]

[image]

kvnper, 2022-12-07:
I can only get the support interface to generate if I change the support **top z distance** to 0:

[image]

[image]
[image]

kvnper, 2022-12-07:
I have now seen that this problem of no interface occurs when the wall angle is greater than 45 degrees from horizontal.

kvnper, 2022-12-07:
I have now seen that this problem of no interface occurs when the wall angle is greater than 35 degrees from horizontal.

[image]
[image]

MusangKingBambulab, 2023-03-23:
hi @kvnper, sorry for my belated reply. Please let me have a brief description of the concept in support：
Interface layer - a more uniform layer between the printed and the support material which can be used as a third filament for easy removal from the printed.
Then, let's go through your question one by one.

For your first concern about the gap between the support material and the object which is intentional and helpful for removing support. If you mind this gap, you can fill it by setting Top Z distance and  Support/object xy distance equal to zero at the same time as shown below. You will see there is no gap but filled with the as-mentioned interface layers.
[image]

For your second concern about the printing quality on the overhang area, the conical surfaces. It is hard to say that all faults are blamed on the gap since we have to make support removable from the printed.

The third question is about the overhang angle. If you set the overhang angle very large, which means support would generate even if there is a small overhang angle.

If you have further questions, please feel free to let me know.

kvnper, 2024-02-29:
> hi @kvnper, sorry for my belated reply. Please let me have a brief description of the concept in support： Interface layer - a more uniform layer between the printed and the support material which can be used as a third filament for easy removal from the printed. Then, let's go through your question one by one.
> 
> For your first concern about the gap between the support material and the object which is intentional and helpful for removing support. If you mind this gap, you can fill it by setting Top Z distance and Support/object xy distance equal to zero at the same time as shown below. You will see there is no gap but filled with the as-mentioned interface layers. [image]
> 
> For your second concern about the printing quality on the overhang area, the conical surfaces. It is hard to say that all faults are blamed on the gap since we have to make support removable from the printed. A support W filament will be a better choice for improving the print quality on the overhang. https://store.bambulab.com/products/support-for-pla
> 
> The third question is about the overhang angle. If you set the overhang angle very large, which means support would generate even if there is a small overhang angle.
> 
> If you have further questions, please feel free to let me know.

Hi Musang,

I believe that the issue is that the support Z distance is the same as the support X/Y distance. There should be another option to set Z-distance like in Ultimaker's Cura slicer. More than one year later and I am still having this issue that I can solve in Cura with the support Z-distance settings.