bambulab/BambuStudio #782: Angled Overhangs are not fully supported by support interface

kvnper, 2022-12-07:
Bambu Studio Version: 1.3
OS: Windows 10

Project file is attached:
[BBS support interface gaps v1.zip](https://github.com/bambulab/BambuStudio/files/10173744/BBS.support.interface.gaps.v1.zip)

When using a support interface, angled walls are not fully supported by the support interface. There are gaps that leave an undesired finish on the underside of the angled wall.

These gaps are larger the higher the angle from horizontal.

[image]
Close-up:
[image]

Conic model:
[image]

This is an example of the result, this print used PVA support for the interface and with 0 gap. Yet because of these gaps there are some unwanted inconsistencies on the angled wall.
[image]
[image]

tangketan, 2023-02-24:
Thanks for the reporting. It's not a bug, but rather a feature request. We'll try to improve.

tangketan, 2023-02-24:
To be clear, this phenomenon only occurs on normal support + grid style. It's related to the rasterization of the grid.
For now, you can use normal support + snug style, which will give you uniform support interface.