bambulab/BambuStudio #788: [feature request] Exclude items during print

BambuLabDiscord, 2022-12-08:
Request that origin from Klipper and very helpfull to reduce waste or aborting a longer print with many items. If one print fail you could exclude that while printing rest. Perhaps in combe with spaghetti detector that let you choose what is failing and exclude this. Pause and able to remove spaghetti or failed print then continue print.

AbeFM, 2022-12-08:
Given the frequency of false errors with S.D., it should be a manual process - although remote would work.

While not as flexible as per-object (which I've seen listed in some g-code, so per object should be possible at print time), a simple bounding box to stop printing might also work.

RidoKilos, 2023-01-05:
Commenting for visibility. Just learned that this was a feature in Klipper and Octoprint. I would absolutely LOVE to have this for my Bambu machines. It would be a boon to any high volume/batch printing user.

clairmont32, 2023-01-25:
Also commenting for demonstrating increased interest (there are apparently 3 dupe issues though). Excluding spaghetti/failed objects would save sooo many print jobs.

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you every one for this good suggestion!

closed 2023-08-02 by SaltWei

codeframe78, 2023-08-02:
Any chance of this making it's way over to the P1P

SaltWei, 2023-08-03:
> Any chance of this making it's way over to the P1P

P1P firmware and APP is under developing by other team. We have transferred this request to them with high priority.

AbeFM, 2023-08-03:
A related, but different topic: I'll often print "one object at a time",
which is great when I want to set up a few objects of different materials
to print while machine is unattended.

Not a huge need, but it would be nice to be able to skip-and-go-to-next,
which might not be as hard to do for one-material-per-object prints as
described.

CoreyAFraser, 2023-08-30:
> > Any chance of this making it's way over to the P1P
> 
> P1P firmware and APP is under developing by other team. We have transferred this request to them with high priority.

Is there an public issue tracker for that team so we can see when there is an update?