bambulab/BambuStudio #791: ability to calibrate multiple filaments and save filament calibrations in the printer/bambustudio

tsdexter, 2022-12-08:
**Is your feature request related to a problem? Please describe.**
related to #669 - when printing with two different brands, types (matte vs. silk), of the same material the flow calibration is often only good for one of the filaments causing issues in the parts of the print with the other filament

**Describe the solution you'd like**
1. Add the ability for the printer to calibrate multiple filaments on different parts of the bed during a multi-filament print
2. Add the ability for the printer to save many filament calibration settings on the printer (or bambustudio) just like filament profiles. For example, after a calibration is complete a popup says "would you like to save this calibration setting" and then you can enter a name like "eSun PLA+ white" and when loading the AMS you can select which saved calibration to use. 
3. Add the ability to save the calibrations in the cloud so that they can be used with P1P and selected when loading filament

**Describe alternatives you've considered**
Currently the only alternative is to print the colors/filaments separately which doesn't always work depending on the model or to add other items to the print bed to get it to calibrate for the filament you want to look the best

**Additional context**
will add some posts from the facebook group that suffer from this problem

rendermani, 2022-12-08:
Yes, I would support all 3  features. All are very usefull .

AbeFM, 2022-12-08:
Also, if you could get to the calibrations, you could alter them. NinjaFlex prints with a lot of stringing, as do most TPU's. A manual override seems better just adding miles of retraction. Especially if you could take the measured/calculated value from the line scanner and apply a percentage override.