bambulab/BambuStudio #793: [macOS] Unable to build

GiaPag, 2022-12-08:
Hi, I'm unable to build on macOS Ventura
``` 
OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.
```
i read and followed your instructions, i also installed openvdb with brew but without success.
Anyone can  help me?
Thanks in advance

lanewei120, 2022-12-09:
@GiaPag 
1. have you built deps before?
2. if yes, are all the deps built successfully on Ventura?
3. could you attach the whole build logs and commands here
thanks

GiaPag, 2022-12-09:
@lanewei120 
Hi, yes
Here you go the log..
[deps.txt.zip](https://github.com/bambulab/BambuStudio/files/10193669/deps.txt.zip)
[build_error.txt.zip](https://github.com/bambulab/BambuStudio/files/10193670/build_error.txt.zip)

Thank you

lanewei120, 2022-12-09:
there is building errors for the deps
[image]

it seems the zstd is missed?

GiaPag, 2022-12-09:
```
giapag: ~ $ ➜ brew install  zstd 
Warning: zstd 1.5.2 is already installed and up-to-date.
To reinstall 1.5.2, run:
  brew reinstall zstd
````

GiaPag, 2022-12-10:
@lanewei120 
Hi
I building successfully deps using 
`export LIBRARY_PATH=/usr/local/Cellar/zstd/1.5.2/lib/ `
I think -lzstd doesn't work.
After i building BabmbuStudio correctly, the same with cmake abd Xcode command.

closed 2022-12-10 by GiaPag