bambulab/BambuStudio #794: Painting leaves gaps on where slopes meet

AbeFM, 2022-12-08:
**Describe the bug**
A clear and concise description of what the bug is.
I painted my object fully, then used gap fill and then went over crack again with sphere tool. When I slice, I get gaps where the two slopes meet.
[image]
If I set the slicing finer, I keep the gap, when I go to the largest (0.28z), most of the gap is gone.

Painted object:
[image]

I suspect it's a rounding error. I feel that, in such cases, the color should protrude 1 extra layer into the part. The situation is the same if I flip the part over (valleys facing upwards).

The part comes out (mostly) find if the valley faces along x/y, but not Z.
[image]

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.
There is no upload button, and I got an error message specifically stating that .3mf isn't supported. It's a weird instruction, considering.

The part itself needed repair before slicing. I did it, but perhaps a clean part wouldn't have the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]
win10, latest version

SaltWei, 2022-12-09:
> There is no upload button, and I got an error message specifically stating that .3mf isn't supported. It's a weird instruction, considering.

3MF file can be compressed into.zip file. Then github can accept the file format.
Will write this operation into instruction.

AbeFM, 2022-12-09:
[CapWithPaintingError.zip](https://github.com/bambulab/BambuStudio/files/10198630/CapWithPaintingError.zip)
Thanks

zhimin-zeng-bambulab, 2022-12-28:
This gap is caused by the precision of slice layer height.

AbeFM, 2022-12-28:
Yeah, not so much by the precision, but rather, if the seam falls on an even integer-ish multiple of the layer height.

But it makes for bad looking painted parts, therefor I feel either expose the edge itself as a line that can be painted (complex), or just round up somehow when two surfaces meet.

Pretty obvious I'm no programmer. :-)

Anyway, a way to keep the other color from poking through would increase the attractiveness/functionality of parts. I'd like to not have the layer height designated by the shortcomings of the painting tool.

Is it possible that painting with "circle" or "flood" instead of "sphere" is to blame?

zhimin-zeng-bambulab, 2023-01-09:
Your idea is very good, and we will try our best to improve it in the future.
But judging from the current code implementation, it is difficult to solve this problem.

AbeFM, 2023-01-11:
If I have any ideas I'll be certain to share. Perhaps turning up/on gap fill or checking the layer above. Maybe run the gap fill detection routine 1/2 layer higher than the layer itself? I can't say I have any great ideas.

Thanks for noting that the concern is logged, I appreciate the feedback.