bambulab/BambuStudio #799: Linux Crash: Bambu Studio SIGABRT when Linux awakes from sleep state

deftdawg, 2022-12-09:
[Relates to #12 - this is a lower severity issue]

When my KDE Neon/Ubuntu 22.04 awakes from sleep Bambu Studio crashes.   Only tested on 1 machine, consistently occurs.

Attached is the GDB backtrace:
```
[Thread 0x7fff10ad6640 (LWP 183215) exited]
[New Thread 0x7fffe7912640 (LWP 183257)]
1. lo    : 127.0.0.1

*** buffer overflow detected ***: terminated

Thread 83 "bambustu_main" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff889d6640 (LWP 170462)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140735485404736) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: No such file or directory.
(gdb) #0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140735485404736) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140735485404736) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140735485404736, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff11ba476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff11a07f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff12016f6 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff1353943 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007ffff12ae76a in __GI___fortify_fail (msg=msg@entry=0x7ffff13538e9 "buffer overflow detected") at ./debug/fortify_fail.c:26
#7  0x00007ffff12ad0c6 in __GI___chk_fail () at ./debug/chk_fail.c:28
#8  0x00007ffff12ae6ab in __fdelt_chk (d=<optimized out>) at ./debug/fdelt_chk.c:25
#9  0x00007fffd7b5b6f7 in BBL::SsdpDiscovery::ssdp_thread() () at /home/deftdawg/.config/BambuStudio/plugins/libbambu_networking.so
#10 0x00007fffd7e134eb in thread_proxy () at /home/deftdawg/.config/BambuStudio/plugins/libbambu_networking.so
#11 0x00007ffff120cb43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#12 0x00007ffff129ea00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) A debugging session is active.

        Inferior 1 [process 170306] will be killed.

Quit anyway? (y or n) [answered Y; input not from terminal]
```

StoneLiBambulab, 2023-02-07:
Hi @deftdawg ,
may be the same as https://github.com/bambulab/BambuStudio/issues/1148

Could you help to verify the libbambu_networking.so lib in this issue.

Thanks!

Stone Li

SaltWei, 2023-03-02:
V1.5.0 public beta has been released and fix shoule be included. Please let us known if still exists. Close at the moment.
Thank you very much!

closed 2023-03-02 by SaltWei