bambulab/BambuStudio #802: Unprintable 3DMF, slicing freezes at 5%

Kraminator, 2022-12-10:
I have a 3DMF file with quite a few objects for 2-color printing, but for some reason, it does not slice. Even after 24 hours on my iMac Studio M1, it is not going more than 5%.

I gladly send the file over, however, due to copyrights, I can not just post it on Github. Please send me an email address or other way to send the file to you.

SaltWei, 2022-12-13:
Thank you very much for reporting issue to us to keep software being improved!
You can send file to email sallt.wei@bambulab.com and we will try to reproduce.

theogott, 2022-12-13:
Is it [sallt.wei@bambulab.com](mailto:sallt.wei@bambulab.com) or 
[salt.wei@bambulab.com](mailto:sallt.wei@bambulab.com)?

SaltWei, 2022-12-13:
Sorry. Typo!
Should be salt.wei@bambulab.com

theogott, 2022-12-13:
For Size reasons, i generally recommend to zip the File and just drag it here into the Field. 
Because Mail only accepts small files.

Kraminator, 2022-12-13:
Thank you, just sent it.

SaltWei, 2022-12-14:
The software is stuck in the base software library clipper. Start to investigate.

Kraminator, 2023-01-19:
Is there any news on this?

QingZhangBambu, 2023-01-19:
> Is there any news on this?

It is a bug on clipper lib. A small wrong data caused that. a little bit tricky. because it's difficult to catch wrong data. I am working hard on it.

QingZhangBambu, 2023-04-06:
We've fixed it. You can get it in the next version.

Kraminator, 2023-04-06:
Great, thank you. When will the new version be released?

[Reto Kramer](https://spikenow.com/r/a/?ref=spike-organic-signature&_ts=20oc0u)	[20oc0u]
Best regards

Reto Kramer
www.techwithkramer.com

QingZhangBambu, 2023-04-07:
in a week or two

SaltWei, 2023-04-18:
https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65 has been published

closed 2023-04-18 by SaltWei