bambulab/BambuStudio #805: Layer number when printing

bigben4001, 2022-12-11:
I would like an option to see what layer the print is on as well as the total number of layers in the print so that users can have a better idea of how far along the print is as well as so that when multicolor prints are running, especially those with more striations than full color, it can be known where the color changes are easier since it can't be determined at the moment where the printer will change or stop changing.

TheCableGuy99, 2022-12-11:
Already posted and acknowledged it will be implemented in the next version

TheCableGuy99, 2022-12-16:
shame this wasn't implement as promised

MadMax389, 2022-12-16:
Might need the coming firmware update to be functional?

spuiuk, 2023-01-25:
I am running a multi colour print (no AMS) using pauses at layers. Seeing a current layer number/total layer numbers in place or along with the percentage completed would be very useful for me.

spuiuk, 2023-01-25:
I noticed that this feature has been added in  v1.4.2. Point #3
https://github.com/bambulab/BambuStudio/releases/

However it mentions X1/X1C and will rely on a future firmware update. I use a P1P and would like to see this feature included there too.

spuiuk, 2023-01-25:
duplicate of #604 ?

timothyzalusky, 2023-02-07:
[https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history

Added support for displaying the current layer number/total number of layers during printing](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history#:~:text=of%20Bambu%20Studio-,Added%20support%20for%20displaying%20the%20current%20layer%20number/total%20number%20of%20layers%20during%20printing,-Added%20secondary%20confirmation)

Issue can be closed

closed 2023-03-02 by SaltWei

Hologos, 2023-03-12:
@SaltWei: Can we get this feature on P1P as well please?

StoneLiBambulab, 2023-03-13:
Hi @Hologos ,
    The layer number can be displayed after the P1P printer firmware is upgraded. 

Thanks!
Stone Li

Hologos, 2023-03-13:
> Hi @Hologos ,
> 
>     The layer number can be displayed after the P1P printer firmware is upgraded. 
> 
> 
> 
> Thanks!
> 
> Stone Li

I'm on latest firmware but there is no layer number information. It works only for X1(C).

Hologos, 2023-03-13:
Here is a proof

[image]
[image]
[image]

StoneLiBambulab, 2023-04-11:
@Hologos , you can report a new one.

Thanks!
Stone Li