bambulab/BambuStudio #808: Touch screen feature / cancel confirmation

werewolfenn, 2022-12-11:
I was doing a print and ran out of filament. I was near the machine and saw the notice come up. Went to hit continue to advance to the next screen and then hit resume. My fat finger hit stop instead, canceling the print. Sadly it was turning out beautifully.
I think it would be nice if the resume option was in that box instead of having to go elsewhere where there's a chance of canceling it rather easily.  Or, if there was a second, confirmation button to be sure that you actually plan to cancel.

randyre, 2022-12-12:
Link with #792

timothyzalusky, 2023-02-07:
[Added secondary confirmation when stopping a printing job in a pop up window.](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history#:~:text=%C2%B6-,Added%20secondary%20confirmation%20when%20stopping%20a%20printing%20job%20in%20a%20pop%20up%20window.,-When%20a%20user)

Issue can be closed

closed 2023-03-02 by SaltWei