bambulab/BambuStudio #810: Entering "10000" in the Scaling Dialog will lead to 2444331 instead of 10000

theogott, 2022-12-12:
**Describe the bug**
I tried to load a very small STL-File and entered 10000% in the Scaling Dialog, see below.
Anytime i enter "10000" and press RETURN, the Program will change the scaling to 2444331
which is of course not useful.

**3mf File for This Bug**
I added the STL-File here as ZIP.

**To Reproduce**
Steps to reproduce the behavior:
1. Drag the STL-File into the Program
2. When the Dialog comes that offers you to "automatically scale it" say "NO"
3. Now enter manually a scale of 10.000% (1e4) into the X-Files
4. See Picture

[image]

**Expected behavior**
If i enter 10000 it should be 10000

**Screenshots**
[Nr-11.zip](https://github.com/bambulab/BambuStudio/files/10203674/Nr-11.zip)

If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
Window 11

SaltWei, 2022-12-13:
We have limited the max value to be 9999%. Please wait the next release and will come soon

theogott, 2022-12-13:
A limitation is, that if i enter "something too high", it will be limited to the MAXVALUE.
Current Limitation is you enter something to High, it will be made to this crazy value 2444331. 
Maybe we can work also on that limiting Algo :-)
Its a small change.

SaltWei, 2022-12-13:
> A limitation is, that if i enter "something too high", it will be limited to the MAXVALUE.

Accept and we will optimize it.
Thank you very much for suggestions!

theogott, 2022-12-13:
I suggest values at least up to 1e5% because sometimes i get models out of Blender in a wrong scaling.
So think about  a technical limitation, and then decide for the "Maxvalue".

LiZ-Li-BBL, 2023-04-21:
The size of model is limited to 9999.99, so iIf you enter a scale that's too large, it will not take effect. But a valid scale input should behave as expected. This issue has been fixed for some time, we are about to close this ticket.

closed 2023-04-21 by LiZ-Li-BBL