bambulab/BambuStudio #811: Stop per plate layer setting from applying to all plates in project

davidjdixon, 2022-12-12:
**Describe the bug**
Layer based switches (eg add pause / add custom gcode) applied to one plate apply are applied to every plate in a multi-plate project

**To Reproduce**
1. Add more than plate to the project and add models to each plate
2. On one plate, right-click a layer on the layer slider and select "pause".
3. Re-slice to verify layer switch/command has been applied
4. Go to another plate with AT LEAST the same number of layers and check the layer slider - the same switch will be shown on this plate.

**Expected behavior**
Per layer changes on a multi build plate should either switch to apply only the plate being worked on, or provide an option to apply to "this plate only"

**Desktop (please complete the following information):**
 - OS: MacOS
 - Ventura

BigJoe1979, 2022-12-31:
This looks to be a duplicate of #369, which has no comments on it either.  

I would also love to see this fix / feature implemented as it makes using the multiple plates feature as-intended impossible where different plates need different pauses.

SaltWei, 2023-03-02:
Already included in v1.5.0 Public beta

closed 2023-03-02 by SaltWei