bambulab/BambuStudio #813: [Feature Request] More profile presets for different nozzle sizes

TheCableGuy99, 2022-12-12:
**Is your feature request related to a problem? Please describe.**
One of the things that make this printer so user friendly and what people love is it prints out the box. You have done a lot of work into setting up amazing profiles that just work! However, you sell different nozzle sizes and when adding these you have only added a single resolution profile for the additional nozzles. .6 is nozzle is supposed to be the best standard with Arachne which you have partly implemented into the slicer. 

**Describe the solution you'd like**
I would like to see more layer height profiles for different nozzles sizes, especially .6 and .2 (even.8 as you sell those as well).

**Describe alternatives you've considered**
Adding my own but I don't know enough about all the settings to make sure it just works.

**Additional context**
Add any other context or screenshots about the feature request here.

SaltWei, 2022-12-13:
Thanks! We are testing more layer height profiles and will be published after being comfirmed.

TheCableGuy99, 2022-12-13:
Okay fair enough, I mean it would be nice to try but it certainly needs more profiles for different nozzles. Supporting not just your own filament but generic as well as you have done so nicely already.

The idea came when I saw this video so I certainly want to try it and see if I can reduce printing times whilst maintaining print quality and layer adhesion (apparently .6 is also better layer adhesion).
https://www.youtube.com/watch?app=desktop&v=WgXM2zPusXo

Thank you for listening :)

SaltWei, 2022-12-13:
You are welcome! Thank you for suggestions to make us keep improving!

jandrop, 2023-01-01:
I think with 0.6 nozzles you can print fast, first, you can avoid the third perimeter, with two perimeters in 0.6 you can get  a nice wall width, the second I you for example print with a 0.12 layer height you can combine the infill every 3 layers because with 0.6 you can get a higher layer height, so comparing the same model with 0.4 nozzles and 0.6 nozzles at the same speed with these two parameters changed (combine infill every 3 layers and 2 perimeters instead three) you will have the fastest print.

QingZhangBambu, 2023-04-21:
more preset profiles had been added

closed 2023-04-21 by QingZhangBambu