bambulab/BambuStudio #814: Delta printer support

hakimio, 2022-12-12:
**Is your feature request related to a problem? Please describe.**
Right now it's not possible to choose a round bed like present in delta 3d printers.

**Describe the solution you'd like**
Option to choose a round bed like in Prusa Slicer.

**Additional context**
I know that "Bambulab" doesn't make any delta printers, but it would be a very nice addition for people who would like to use BambuStudio with non-Bambulab printers.

SaltWei, 2022-12-13:
We don't have so many 3-rd party printers(For example, delta printer, so need to buy one) to check and add new profiles, and also very busy to solve bambu's issue. So this feature request will not be very high priority.

But we welcome the community developer to add 3-rd party printer profile which has been tested into BambuStudio.

hakimio, 2022-12-13:
It's impossible to add a delta printer profile when round beds are not supported.
Anyway, thanks for looking into this.

SaltWei, 2022-12-13:
> It's impossible to add a delta printer profile when round beds are not supported.

Thank you very much for suggestions! Will add it into wish list frist.