bambulab/BambuStudio #815: Additional filament sensor

werewolfenn, 2022-12-12:
**Is your feature request related to a problem? Please describe.**
Several times, filament has gotten tangled at the spool (Backfeed, not AMS). Tension has been so tight, that it stops feeding into the hot end BUT it still triggers the filament sensor that it thinks there's still filament.  It will keep printing for hours and hours until the gcode tells it to quit. 

**Describe the solution you'd like**
Perhaps a second sensor that detects if plastic is extruding?

SaltWei, 2022-12-13:
Feature and issue related to printer should not be posted here .
BTW, we will report this ticket to printer team, but maybe they will not give response here again.
Issue related to printer can be reported by app.
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

SaltWei, 2022-12-13:
OK. Response from printer team.
This is a great idea and they will consider it in the next generation printer.
But for X1 which has been released, it's hard to change now. AMS can detect whether spool is stuck. And user can use AMS to print single color.