bambulab/BambuStudio #817: [Bug] Thumbnails not showing in Bambu Studio

TheCableGuy99, 2022-12-13:
**Describe the bug**
After uploading the sliced file to the printer it switches automatically to the "Device" tab. At the bottom under the camera view there is the "Printing Process" and a square which looks like the thumbnail should be viewable but it's not. It's just a generic image to signify there is no thumbnail.

**3mf File for This Bug**
It is all files

**To Reproduce**
Steps to reproduce the behavior:
1. Slice File
2. Upload to printer
3. Go to device tab
4. See printing process image/thumbnail

**Expected behavior**
A thumbnail of the object being printed to appear

**Screenshots**
Attached

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro 64bit (latest version with all updates installed)
 - Version: 22H2
 - OS Build: 22621.900
 - Experience: Windows Feature Experience Pack 1000.22638.1000.0
[image]

TheCableGuy99, 2022-12-13:
As an update to this I wanted to note that I have just sliced 2 more files on the slicer for 2 different printers and the thumbnail is working. I think it needs more investigation as to why it's not working sometimes. I will keep an eye on it and see if I can locate the issue.

TheCableGuy99, 2022-12-13:
Further update, I believe this is tied into another bug in where it loses connection to the printer and you need to re-connect. After this happens the thumbnail disappears.

Obviously this is down to you how to resolve, I know you are aware of the issue with it losing connection to the printer so once that is resolved it will resolve this issue, however, may I suggest it stores the thumbnail in the software so it can be recalled if connection to the printer is lost? It's not essential, but it's nice to see the thumbnail.

Thanks.

StoneLiBambulab, 2023-01-09:
Hi，
     Thumbnail will not display when start a print job from printer storage. This feature is not yet implemented. So we display a place holder thumbnail.
     we plan to optimized in a recent release.

Thanks!

Stone Li

StoneLiBambu, 2024-01-29:
@sodachen22 , 
Whether there is a plan to support pulling thumbnails from the printer device?

Stone Li（旺仔）