bambulab/BambuStudio #818: Ability to add fonts

JasonSTX, 2022-12-14:
It would be great if we could add fonts to the 3D Text tool.  Also, add the ability to bend and shape the text around objects.  It doesn't matter if we need to manually specify the curve radius.

StoneLiBambulab, 2023-03-02:
Attempt to install fonts to your system, the fonts are loaded from the operating system's font library.

Thanks!
Stone Li

JasonSTX, 2023-03-02:
> Attempt to install fonts to your system, the fonts are loaded from the operating system's font library.
I tried on a fresh install.  The font is installed and usable in other apps but not Bambu.

suchabadperson, 2023-03-05:
@StoneLiBambulab Same issue here, not every font is visible and usable in the text tool. I am unable to use a font recently added to the system, in both Windows and macOS.

Photogad, 2023-03-27:
+1 we need ability to use other fonts. @LiZ-Li-BBL it seems the program DOES use some fonts the user has installed on the system, but not others. I do notice it only displays fonts that are TrueType files (not just .ttf files, but actual TrueType files) AND digitally signed by the creator. OpenText format does not seem to be suppoted at all, even if it's using TrueType outlines.

suchabadperson, 2023-05-08:
Any update on this? @LiZ-Li-BBL

JeffreyBrew, 2023-08-31:
Something I just discovered that worked for a Comfortaa font I downloaded from Google (https://fonts.google.com) but did not work for the Arial Rounded MT Bold font that was installed by Microsoft 365 (M365, formerly Office365).  I discovered there are two ways to install fonts in Windows. If you right-click on the font file you want to install you will be presented with two install options... 'Install' and 'Install for all users'.  If you are using Windows 11, only the 'Install' option is shown on the first context menu.  You will have to select 'Show more options' on the first context menu to get to the secondary context menu that has both install options.  (see pics, below)

When I installed the Comfortaa font with the 'Install' option, I was not able to see the Comfortaa font in Bambu Studio (BS).  After deleting the Comfortaa font file then re-installing it with the 'Install for all users' option, I was able to see the Comfortaa font in BS.  Unfortunately, this did not work for the Arial Rounded MT Bold font that was installed by M365, which is the font I really to access in BS.

Hopefully, this workaround will work for you too.  This is not an issue with Prusa slicer which might be a more viable workaround.  A basic printer profile for the x1C can be downloaded from Printables.com (https://www.printables.com/model/272378-prusaslicer-profile-for-bambu-lab-x1-carbon-04mm-b) for Prusa Slicer.

Finally, I'd like to thank Bambu for their support, or lack thereof, fixing this issue in BS.

First context menu
[image]

Secondary context menu
[image]

toddanglin, 2023-11-13:
@LiZ-Li-BBL Any updates on this issue?

I'm also trying to get more of the fonts available on my macOS machine to show in BBS. At a minimum, it would be very helpful if you could share more details about how BBS is loading fonts today:

- Where is it looking for fonts to load (on Windows/macOS)?
- What conditions must apply for a font to work (file type, etc)?

Hoping to see BBS gain parity with PrusaSlicer on this feature. As is, I have to use a clunky workflow that involves embossing an object in PrusaSlicer then importing the changed object in BBS to slice for Bambu printers. 😬

toddanglin, 2023-11-13:
Ah! With the help of GitHub Copilot I was able to investigate the source code and found answers to my questions! Here are the conditions BBS applies when loading fonts:

1. Fonts that start with "." are ignored. 
2. Any fonts with "Emoji" in their name are ignored. 
3. Fonts containing specific suffixes in their name are ignored. The suffixes indicate weights or variations (like "Bold", "Italic", or "Light") of the fonts.
4. Lastly, if a font's file extension is not ".ttf", ".otf", or ".ttc", it is also excluded. 

Critically, via Copilot, "the `Font_FontMgr` object is initialized using `aFontMgr->InitFontDataBase()`. This operation generally loads the font database **from the system's default locations**."

As it turns out, on macOS, that location is `/Library/Fonts` and **NOT** `~/Library/Fonts` (where you'd most likely find any fonts you've installed on your machine). Put a font that meets the criteria in `/Library/Fonts`, reload BBS, and you should see the font in the Text Shape list! (And, as you might expect, this also makes the fonts available in OrcaSlicer, too.)

JeffreyBrew, 2023-11-13:
@toddanglin,
Thanks for your feedback.  This explains why I am not able to load the _Arial Rounded MT Bold_ (ARLRDBD.TTF) font in Bambu Studio.  Based on your feedback, because this font contains _Bold_ in its name, Bambu Studio ignores it.  Since the font's name is an internal property of the font file, I am not able to change the font name to trick Bambu Studio into loading this font. 
Unfortunatley, there are not other versions of this font file.  There is only _Arial Rounded MT Bold_.

@BambuStudio
Please remove the limitation(s) that prevent Bambu Studio from loading _Arial Rounded MT Bold_.  Specifically, please remove the third limitation @toddanglin mentioned in his message.

toddanglin, 2023-11-14:
For reference, here are all the font file "suffixes" that are being scanned and filtered:
```
static const std::vector<Standard_CString> fonts_suffix{ "Bold",  "Medium", "Heavy", "Italic", "Oblique", "Inclined", "Light", "Thin", 
"Semibold", "ExtraBold", "ExtraBold",  "Semilight", "SemiLight", "ExtraLight", "Extralight",  "Ultralight", 
"Condensed", "Ultra", "Extra", "Expanded", "Extended", "1", "2", "3", "4", "5", "6", "7", "8", "9", "Al Tarikh"};
```
This all happens in the `TextShape` class: [`src/libslic3r/Shape/TextShape.cpp`](https://github.com/bambulab/BambuStudio/blob/693aa8d259f2eaeb30ac1055732ff5f282a2c535/src/libslic3r/Shape/TextShape.cpp#L35C1-L37C114)

And to affirm, yes, the OCCT library is reading the actual font _names_ and not the file name, so, changing the file name is insufficient for avoiding this filter. I tested this with a few font files just to double-check.

FWIW, I reviewed the source for PrusaSlicer, and it seems to have a completely different implementation. Rather than defining a `TextShape.cpp`, PrusaSlicer does all of its work through `Emboss` classes. And [when it loads fonts](https://github.com/prusa3d/PrusaSlicer/blob/87a58670f3a20dbda2478e35f7b74663d36c099b/src/slic3r/Utils/WxFontUtils.cpp), it 1) loads them from the user space font directory (i.e. `~/Library/Fonts`), and 2) it applies minimal filtering (basically just validating the file is a truetype font). Long term, it would be great to see BambuStudio (and/or OrcaSlicer) converge on the PursaSlicer implementation as it does seem more capable and aligned with user expectations. But I'm wading deep into waters where I am no expert. :)

toddanglin, 2023-11-14:
> This explains why I am not able to load the _Arial Rounded MT Bold_ (ARLRDBD.TTF) font in Bambu Studio. Based on your feedback, because this font contains _Bold_ in its name, Bambu Studio ignores it.

@JeffreyBrew In the meantime, as a workaround, you should be able to use a font program to rename your font and thus enable it to load in BambuStudio. I tested this today with a free program called [FontForge](https://fontforge.org/en-US/) and was able to modify a font to make it load in BambuStudio.

Critical values that need to be changed in `Element > Font Info...` (in FontForge):
1. In `PS Names` -> FontName, Weight (just delete the value)
2. In `OS/2` -> Style Map (set it to 'Regular')

YMMV, but hope that helps. And, as before, make sure your modified font lives in the _system_ font directory.

JeffreyBrew, 2023-11-14:
> In the meantime, as a workaround, you should be able to use a font program to rename your font and thus enable it to load in BambuStudio. I tested this today with a free program called [FontForge](https://fontforge.org/en-US/) and was able to modify a font to make it load in Bambu Studio.

@toddanglin, Thanks for the feedback.  You're awesome!  I will definitely check out FontForge to see if I can get the Windows _Arial Rounded MT  Bold_ font to load.  FontForge looks like an interesting piece of software.
FWIW... in an email conversation I had with Bambu support, I was informed the developers had assigned this issue a "low priority" and was not on the development schedule.  Due to time constraints imposed by the USPTO, I couldn't wait for Bambu to get around to fixing this issue.  I was able to find a similar fonts at [Google Fonts](https://fonts.google.com/), install them in Linux, print the part I needed with Prusa Slicer and a brand new Prusa printer.  Prusa made it possible for me to print the "branding specimen" I needed to submit to the USPTO for a trademark I had applied for.  Since then, that trademark has been approved and I am completing a few tasks on my end to get that trademark registered (®).

@BambuStudio
I understand you are trying to make the BS software easier for users to use.  What you are failing to understand is users of 3D printers are a smart group of creative people.  That's why we are labeled as "creators" and "makers".  When you deny us the tools we need (i.e., fonts), you limit our ability to do what we do... create and make... we go elsewhere to get what we need.  For me, Prusa saved the day.  Prusa printers just work.  And, Prusa support is available 24-hours/day.

enewbold, 2023-12-02:
As of Dec 2, 2023, there is still no help or fix from Bambu Labs about this glaring issue!  What's WRONG with those people???

JeffreyBrew, 2023-12-02:
> As of Dec 2, 2023, there is still no help or fix from Bambu Labs about this glaring issue! What's WRONG with those people???

I gave up on Bambu as a company and regret having bought a printer from them.  Bambu support is beyond horrible.

gabriel-visualthinking, 2024-02-01:
Still not fixed...

toddanglin, 2024-02-01:
FWIW...I've now used the workaround I described above several times to add custom fonts to Bambu Studio (and OrcaSlicer). It's cumbersome, but it works, so if you need a custom font short-term, I would follow those steps. Hopefully, support for custom fonts will improve as Studio adopts the entire "Emboss" workflow from Prusa Slicer.

toddanglin, 2024-03-09:
On macOS, make sure the modified font is in the _system_ fonts folder. From before:

> As it turns out, on macOS, that location is /Library/Fonts and NOT ~/Library/Fonts (where you'd most likely find any fonts you've installed on your machine). Put a font that meets the criteria in /Library/Fonts, reload BBS, and you should see the font in the Text Shape list! 

If you've put your font there and you're still not seeing it in Orca/Bambu, the odds are there is some other "offending" value in the original font file that's causing the font to be filtered. In that case, I'd go back through all the values you can see using the FontForge tool and change anything that looks like it has to do with font weight to "normal" values. It may take some guessing and checking.

After that, I'm out of ideas. :)

bolsoncerrado, 2024-03-09:
NOt sure if OS/2 Panose Weight had to do, but changed to ANY instead of BOLD and now it worked ;) Ty!

suchabadperson, 2024-04-23:
Since version 1.9 beta 4, fonts are displayed in the list, but it does not work correctly - the design of the font itself is not displayed but a generic one.

zhimin-zeng-bambulab, 2024-04-24:
@suchabadperson  Some fonts are not valid for Chinese, and the default style will be used.

suchabadperson, 2024-04-24:
@zhimin-zeng-bambulab There are many fonts that are very useful because of their design, although unfortunately they do not have all the characters. Are they not going to be supported? In PrusaSlicer they work correctly.

zhimin-zeng-bambulab, 2024-04-24:
@suchabadperson  Could you list a few representative fonts so that we can study them?

suchabadperson, 2024-04-24:
Here is a screenshot of what I was talking about. It would be nice if Bambu Studio could support any font as PS does. 

[image]

suchabadperson, 2024-05-13:
Issue still present on 1.9.1 Public Official release. Is this being worked on? @zhimin-zeng-bambulab

Henlor, 2024-05-16:
@zhimin-zeng-bambulab
Indeed, it's not working working as intended.
All the fonts seem well loaded inside the dropdown menu. But if a font is not a system font, font installed by the user, it's no applied to the text in the 3d view and shows Arial font instead. 

As is, the text tool is even more confusing and makes the user experience worst than before.

Bambustudio 1.9.1.67
MacOS 13.3.1

[image]

zhimin-zeng-bambulab, 2024-05-16:
Thank you for your feedback. This issue is being processed and we will fix it as soon as possible. Thanks!

otterreading, 2024-06-14:
> Thank you for your feedback. This issue is being processed and we will fix it as soon as possible. Thanks!

Do you have any updates on this? I am trying to use a specific font and it shows up in Bambu Studio, but the actual font isn't correct. Thanks!

Sonicboom247, 2024-07-04:
can we have some forward progress on this? If BS is based on Prusa Slicer and the fonts work as intended there, then it seems like work went into limiting the use of many fonts instead of just leaving it alone.  I keep almost 200 or more fonts on my NAS that I just copied over and installed and although I can see their name, none of them change the font.

suchabadperson, 2024-08-26:
Same issue on 1.9.4.

GerNavBet, 2024-09-23:
Same issue on 1.9.7.52.

Sonicboom247, 2024-09-23:
There is a confirmed workaround posted on Bambu forum.

GerNavBet, 2024-09-23:
Thanks, all sorted and working now!

suchabadperson, 2024-09-23:
> There is a confirmed workaround posted on Bambu forum.

Could you share the link?

bolsoncerrado, 2024-09-23:
+1 for the link!

Sonicboom247, 2024-09-23:
https://forum.bambulab.com/t/adding-custom-fonts-to-bambu-studio/5936

suchabadperson, 2024-09-23:
It’s still not working for me. The fonts are visible in Bambu Studio, but some of them appear as the default when applied to an object.

Same as before here:
https://github.com/bambulab/BambuStudio/issues/818#issuecomment-2075478262

Sonicboom247, 2024-09-23:
Make sure you install each and every one with admin rights, I think I have one or two that just don't show up but for the most part 80 of mine now work

suchabadperson, 2024-09-23:
My issue is on macOS, it might work fine on Windows, need to try.

donnay, 2025-05-07:
using version of BBS - V 2.0.3.54 - Issue remains outstanding. Any updates?