bambulab/BambuStudio #819: Bambu Studio Crashing immediately on Mac OS

r259c280, 2022-12-14:
**Describe the bug**
When attempting to open Bambu Studio on Mac OS, it crashes immediately. 

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to open up the Bambu Studio application
2. See error

**Expected behavior**
The application should load and function as expected. 

**Screenshots**
No screenshots but here is a gist 
https://gist.github.com/r259c280/18\bceaf0f4a73ef99079cb9f3f9d4dc8
**Desktop (please complete the following information):**
 - OS: MAC 
 - Version Ventura 13.0.1

I also logged a ticket: US221213600005
and this can be referenced as well: https://www.reddit.com/r/BambuLab/comments/zl0emn/bambu_studio_keeps_quitting_unexpectedly_on_mac/

lanewei120, 2022-12-14:
> **Describe the bug** When attempting to open Bambu Studio on Mac OS, it crashes immediately.
> 
> **To Reproduce** Steps to reproduce the behavior:
> 
> 1. Attempt to open up the Bambu Studio application
> 2. See error
> 
> **Expected behavior** The application should load and function as expected.
> 
> **Screenshots** No screenshots but here is a gist [https://gist.github.com/r259c280/18\bceaf0f4a73ef99079cb9f3f9d4dc8](https://gist.github.com/r259c280/18%5Cbceaf0f4a73ef99079cb9f3f9d4dc8) **Desktop (please complete the following information):**
> 
> * OS: MAC
> * Version Ventura 13.0.1
> 
> I also logged a ticket: US221213600005 and this can be referenced as well: https://www.reddit.com/r/BambuLab/comments/zl0emn/bambu_studio_keeps_quitting_unexpectedly_on_mac/
> 
> **Workaround** Currently the workaround is using the latest version of https://github.com/SoftFever/BambuStudio-SoftFever/releases
> 
> good find! @3dprinterAcademy

@r259c280 

Could you delete the application totally and install again?
including following things:
delete 'BambuStudio' in the Applications
rm  /Users/user.name/Library/Application\ Support/BambuStudio

lanewei120, 2022-12-31:
this issue is caused by the installing files wrong or mistaken with other installs
not a program issue

closed 2023-01-05 by r259c280