bambulab/BambuStudio #822: Option for using Plate Coordinates when positioning objects on build plate using the Move tool

sjscymru, 2022-12-14:
**Is your feature request related to a problem? Please describe.**
When using multiple build plates I am frustrated when positioning the objects on any of the plates other than the first plate as the object coordinates are always displayed in world coordinates, which makes positioning on the plate more difficult (if for instance you wanted to position the object directly in the middle of the 2nd build plate)

**Describe the solution you'd like**
An option to toggle between "world coordinates" and "plate coordinates" when using the Move tool. Or just use plate coordinates as the default as I'm not sure as to the value of world coordinates.

**Describe alternatives you've considered**
I don't believe there are any alternatives

JonLaliberte, 2025-02-01:
Seriously yes.
I can’t make heads or tails of what the actual coordinate system is.

JonLaliberte, 2026-01-14:
@thegreatco 
Nice!
I was looking at this today and noticed that H2D table has a type for plate 3 it should be: 840,0

I had Claude create a script to generate tables for the h2d of the different plate positions, and it can also translate between plates. For example if you have the objects position on plate X you can get the same position on plate Y.
https://gist.github.com/JonLaliberte/370e599939405a08a9ef4cf7ed2bbbd7