bambulab/BambuStudio #824: [Bug] Failed resuming after filament runout

TheCableGuy99, 2022-12-14:
**Describe the bug**
I was printing some cylinders at standard speed and super-draft quality. It was all going great and then it ran out of filament. I changed it with the exact same filament, fresh from an air-tight sealed wrapper and clicked "retry" on the LCD screen.
I've done this many times and always been amazed at how well it resumes, but this time i checked the camera a few minutes later and saw it wasn't going to plan, something had messed up (possibly with the layer height skipping a layer or two) so had to stop it. I wonder if there's a bug with resuming prints at that layer height?

**3mf File for This Bug**
I was printing all 12 spool holder spindles from this: https://www.printables.com/model/76519-spool-holder/files

**To Reproduce**
Steps to reproduce the behavior:
1. Download all 12 different sizes of the spool_holder_spindle files
2. Put them all on the bed and slice at 15% grid infill and extra-draft layer height. 
3. Run out of filament around 50 of the print
4. Change with exact same filament and resume print by clicking "retry" on the LCD.
5. Check print in a few minutes

**Expected behavior**
To continue the print with the new filament and no issues.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - Windows
 - 11 Pro 64 Bit

TheCableGuy99, 2022-12-14:
I've re-sliced this, re-started with the same layer height and used the same filament. During the print the filament ran out again, so I replaced it with a new sealed roll and clicked "retry" on the LCD just like last time. This time it restarted perfectly as I'd expect. So it's not something to do with restarting with the layer height. 

Something went wrong but i'm not sure what and suppose I'll have to put it down to one of those things.

SaltWei, 2022-12-15:
Seem to be printer or ams related issue, rather than slicer issue. Will transfer to printer team

SaltWei, 2022-12-15:
Printer team need log to analyze. Please report this issue and upload log by APP following this guide.
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

closed 2022-12-16 by TheCableGuy99

michalss, 2023-01-29:
i can confirm this as well, its a BUG for sure..

TheCableGuy99, 2023-01-30:
> i can confirm this as well, its a BUG for sure..

I have to admit I totally forgot about this but it only happened the once and there's been firmware updates since then. I've had plenty of other instances where filament ran out and i've not been able to replicate it so not sure if it's a bug (for me anyway) anymore.