bambulab/BambuStudio #825: Artifacts on print after loading step-file

hutje, 2022-12-14:
**Describe the bug**
Artifacts on the surface after importing STEP-file in BambuStudio

**3mf File for This Bug**
[Strange Artifacts.zip](https://github.com/bambulab/BambuStudio/files/10231121/Strange.Artifacts.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Import step-file in BambuStudio
2. To see more clearly, open paint supports
3. Select bucket-tool with 100% smart fill angle
4. Hover over the surface and see the strange artifacts

**Expected behavior**
To compare it with how it should look, I have attached a high quality mesh of the same object.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version  BambuStudio V01.03.01.01

SaltWei, 2022-12-16:
Seems to be bug of Opencascad library. Will investigate further

SaltWei, 2023-02-09:
This is stepToMesh format conversion issue of OpenCasCad library and not easy to be fixed.

We decided to increase the precision of conversion so this defect can be smaller in visually when in prepare page.
[image]

By the way, when precision of conversion is increased, the final gcode also can be smooth as arc when arc fitting is enabled. And the defect is also unvisiable in final printing model.

Increasing the precision of conversion will be included in next release.
Thank you very much for suggestions!

SaltWei, 2023-03-02:
> Increasing the precision of conversion will be included in next release.

V1.5.0 public beta has been released

closed 2023-03-02 by SaltWei

amzaldua, 2024-06-03:
Red STL
Yellow STEP

[image]
[image]