bambulab/BambuStudio #836: Internal Solid Infill / Ensure vertical thickness

Bpositivedotdk, 2022-12-15:
The new "ensure vertical thickness" option does not work.  When unticked it's supposed not to make Internal Solid Infill at the sloping walls, but still does. 

This is at 2 walls of 0,6. At 3x walls it doesn't add it. 

For my particular model is ads 32% of unneeded print time. 

Version 1.4.0.17

[image]

SaltWei, 2022-12-16:
Could you please share the 3mf file for us to investigate?

FlowForm, 2022-12-17:
I have send you a mail with the 3mf :-)

SaltWei, 2022-12-22:
Sorry, I haven't received it. Did you send to salt.wei@bambulab.com?

SaltWei, 2023-02-08:
Sorry, I haven't received the 3MF file. So I just use a ball to check but find no issue.
[image]

SaltWei, 2023-03-02:
Seems no update for a long time. Please let us known if still has issue and we will reopen.
Close at the moment.

closed 2023-03-02 by SaltWei