bambulab/BambuStudio #841: [Bug] AMS not changing between printers with new remember function

TheCableGuy99, 2022-12-16:
**Describe the bug**
One of the listed improvements is:
3. Remember the filament list and flush volumes:
https://github.com/bambulab/BambuStudio/issues/452 
https://github.com/bambulab/BambuStudio/issues/698

However, when switching between multiple printers it retains the old printers AMS's, filaments and flushing volumes.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Sync printer with AMS
2. Flush Volumes
3. Switch to second printer
4. Sync with AMS
5. Flush volumes
6. Switch back to previous printer and previous printers AMS configuration is loaded instead of the current printer

**Expected behavior**
When switching between printers I would expect it to query each printer to update the AMS and filaments connected and loaded. An option to also have it flush the volumes would be nice if you want to but that's a feature request I believe unless it's already supposed to be an option?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Windows 11
 - 1.4.0.17

lanewei120, 2023-03-02:
the same issue as https://github.com/bambulab/BambuStudio/issues/1122

bambu123, 2023-03-08:
Please use the methods mentioned in #1122 to cover this problem currently.

closed 2023-03-08 by bambu123