bambulab/BambuStudio #842: Unable to select items in new slice and print drop down boxes

FreeManFreeThinker, 2022-12-16:
**Describe the bug**
I am unable to select any of the options in the new drop-down menus for slice and print. As soon as my cursor is on the drop-down box, it disappears. 

**3mf File for This Bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Open a file
2. Click on the small arrow pointing downwards next to other slice or print
3. Try and click an option

**Expected behavior**
The drop-down box stays open and the action I click gets performed.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Apple Silicon - Ventura 13.1

FreeManFreeThinker, 2022-12-16:
Fully uninstalling and reinstalling using AppCleaner.app fixed it for me

talldonkey, 2022-12-23:
I am also encountering this exact issue.
macOS, Apple M1 Pro, Ventura 13.0.1

walterwongbbl, 2023-01-11:
> Apple Intel Core, Venture 13.0.1. I have the same Issue with 1.4.0.18

Please wait for the next version update.  We have fixed this issue.

SaltWei, 2023-03-02:
v1.5.0 public beta has been released as well

closed 2023-03-02 by SaltWei