bambulab/BambuStudio #843: Change Filaments at Layer no longer functions

Llorean, 2022-12-16:
**Describe the bug**
The feature to add a filament / color change at a specific layer no longer works properly as of 1.4.0.17. Previously, when you clicked "print" you could assign AMS filaments to color 1 and color 2. Now, when I've selected to have a filament change at a specific layer, when I tell it to print, there's only one filament/color displayed as an option, so it can only be printed as a single solid color with no layer-based filament change.

**To Reproduce**
1. Load STL file.
2. Click "+" to add a second filament.
3. Assign filament 1 to an AMS slot
4. Assign filament 2 to a different AMS slot
5. Slice
6. Scroll to layer chosen
7. Right click on the layer indicator and choose "Change filament" and select filament 2.
8. Slice again and verify the preview shows 2 colors.
9. Click "Print plate."
10. Observe that only one color is visible and available in the preview window, and you cannot assign 2 different AMS slots here.
(If printed, it comes out single-color).

**Expected behavior**
At step 10, two filament slots should be shown that can be assigned slots from the AMS.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

Example 3MF:
[Example.zip](https://github.com/bambulab/BambuStudio/files/10242140/Example.zip)

SaltWei, 2022-12-16:
Yes, we can reproduce this issue and are investigating now.
Currently, you can try to use SD card to print as workaround.
Thank you very much for reporting this issue to us！

Manfredo1952, 2023-01-06:
Just tried to do that filament change at a certain layer with my system here (iMac, Studio 1.4.1 and X1C/AMS latest FW) and it does work well.   Had a cube sliced, moved slider (right side) to a layer,  right click on the +, sleect change filament, select colour from AMS, etc. Works pretty well.  
[image]

Manfredo1952, 2023-01-06:
@SaltWei   Hi, this reported bug can be assigned as closed/solved, no ?   see # 4 of latest Studio version 
[image]

Llorean, 2023-01-06:
Which version are those bug fixes on? The current distributed version via builtin updates, 1.4.0.18 still shows the issue, at least on Windows 11.

Llorean, 2023-01-06:
Ah, missed the 1.14.1. It appears that version hasn't rolled out via the auto-update, at least at my end. So, fixed in a version marked as "beta."

StoneLiBambulab, 2023-01-16:
@Llorean ,
   Hi, I fixed the issue and the hotfix version will be published this week.

Thanks!

Stone Li

SaltWei, 2023-03-02:
v1.5.0 public beta has been released

closed 2023-03-02 by SaltWei