bambulab/BambuStudio #844: Mac: Dark mode is inconsistently applied

Fire-, 2022-12-16:
> **Describe the bug**
> A clear and concise description of what the bug is.

The dark mode theme ( which was automatically enabled when I updated, and I cannot locate a setting to disable ) is inconsistently applied across the application, resulting in unreadable text in various locations, such as the Prepare tab's Objects list, and various labels in the Prepare and Project tabs.

> **To Reproduce**
> Steps to reproduce the behavior:

1. Open the app?

> **Expected behavior**
> A clear and concise description of what you expected to happen.

The theming should be applied more consistently across the UI, so that there aren't dark grey text elements on top of even darker grey background elements.

> **Screenshots**
> If applicable, add screenshots to help explain your problem.

[image]
[image]

> **Desktop (please complete the following information):**

 - OS: Mac OS 13.1
 - Version 1.4.0.17

walterwongbbl, 2022-12-16:
Hi.
Will the other pages do the same.  how about device page?

walterwongbbl, 2022-12-16:
> > **Describe the bug**
> > A clear and concise description of what the bug is.
> 
> The dark mode theme ( which was automatically enabled when I updated, and I cannot locate a setting to disable ) is inconsistently applied across the application, resulting in unreadable text in various locations, such as the Prepare tab's Objects list, and various labels in the Prepare and Project tabs. After additional testing, this extends to light mode as well, almost like these elements have their associated theme colors inverted.
> 
> > **To Reproduce**
> > Steps to reproduce the behavior:
> 
> 1. Open the app?
> 
> > **Expected behavior**
> > A clear and concise description of what you expected to happen.
> 
> The theming should be applied more consistently across the UI, so that there aren't dark grey text elements on top of even darker grey background elements.
> 
> > **Screenshots**
> > If applicable, add screenshots to help explain your problem.
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> > **Desktop (please complete the following information):**
> 
> * OS: Mac OS 13.1
> * Version 1.4.0.17
> 
> For added context, I've only been able to change the app's theme thus far by changing the entire system theme, which is a fairly undesirable association.

walterwongbbl, 2022-12-16:
@Fire- 
Hi.
Will the other pages do the same. how about device page?

Fire-, 2022-12-20:
@walterwongbbl the Device page seems to look fine

[image]

walterwongbbl, 2022-12-21:
Hi.
Have you changed the font or color of the mac system.

Fire-, 2022-12-21:
> Hi. Have you changed the font or color of the mac system.

no, they're all default

walterwongbbl, 2023-01-09:
This will be fix this issue in the next version

SaltWei, 2023-03-02:
v1.5.0 public beta has been released. Please have a try, thanks

closed 2023-03-02 by SaltWei

Fire-, 2023-05-08:
@SaltWei checking in ( apologies for how late this is ) with 1.6.24 it appears that many of the text visibility issues have been corrected, however, many still remain; this ranges from "difficult" to "impossible" in terms of legibility

[image]
[image]

walterwongbbl, 2023-05-09:
Hi.
Thank you for your feedback. We will optimize the color of the text in dark mode