bambulab/BambuStudio #845: Timelapse is ALWAYS enabled

Max-Plastix, 2022-12-16:
**Describe the bug**
The new checkbox at Print time to select/deselect Timelapse is broken.  It is not possible to disable the timelapse recording.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Slice a file.
2. "Print plate"
3. Deselect "Timelapse" in the "Send print job to..." dialog box
4. Notice Timelapse file is created on the SD Card.

**Expected behavior**
I would expect the GCODE for timelapse to be removed, and the print to not generate a timelapse file on the card.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Windows 10
 - Version 1.4.0.17

YifanWuBambu, 2022-12-16:
This will be supported in new printer firmware v.1.3.0.0, which will be released soon.
We should mark this in the release note, sorry for that.

Max-Plastix, 2022-12-16:
My X1c & AMS upgraded firmware today (1.03.00.00 & 00.00.05.75), and this problem is completely resolved.

I printed Vase mode with and without timelapse, and it correctly recorded and did-not-record.
I'm closing this out as a mismatch of New Studio with Old Firmware.

closed 2022-12-16 by Max-Plastix

bakaufman, 2023-01-16:
I am running 1.4.0.18 and there is no way to turn timelapse off in others. So I have to uncheck the box on Every print, which is ultra annoying.