bambulab/BambuStudio #846: New 1.4.0.17 Timelapse gcode introduces pauses & defects in Vase mode, always.

Max-Plastix, 2022-12-16:
**Describe the bug**
The new Bambu Studio v1.4.0.17 moves the Timelapse selection to Print time instead of Slice option.
I am printing a Vase-mode cylinder, and have Disabled the Timelapse option at print time.
The vase prints with a visible anomaly, like a seam, caused by a noticeable pause in the motion of the print head.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.
(Any vase mode print)

This inserts the gcode to take timelapse photos at each layer:
```
; CHANGE_LAYER
; Z_HEIGHT: 8.8
; LAYER_HEIGHT: 0.2
G1 Z8.600 F28800
; layer num/total_layer_count: 44/250
M622.1 S1 ; for prev firware, default turned on
M1002 judge_flag timelapse_record_flag
M622 J1
 ; timelapse without wipe tower
M971 S11 C10 O0
M623
```

**To Reproduce**
Steps to reproduce the behavior:
1. Load or create a Vase-mode print.. cylinder with no top or bottom, for example.
2. Enable Vase mode, of course.
3. Slice the file
4 Send to printer, deselecting the Timelapse option.
5. Witness the pauses in each layer, and the visible scar on the printed vase.

**Expected behavior**
Previous versions of Bambu Studio would print a Vase smoothly, without pauses or blobs, when Timelapse was not enabled.

The new version does not have the option to print without Timelapse insertions, and always stutters when printing each layer, unlike the previous version.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Bambu Studio v1.4.0.17

SaltWei, 2022-12-16:
Thank you very much for reporting issue to us. Do you mean this issue is reproduced even you are using traditional timelapse?

SaltWei, 2022-12-16:
See the latest message of this ticket([https://github.com/bambulab/BambuStudio/issues/753](url)), and I can't reproduce this issue on vase mode with traditional timelapse. 3MF and videos are attached.

Could you update one short video of pause during printing for us to look. Just we guess that maybe the pause is caused by printer bug when take photo.

SaltWei, 2022-12-16:
OK. The difference between our printers is my printer is using internal new firmware and your printer firmware should be old release version. Maybe this is the cause.

Printer firmware will be updated in recent several days.

Max-Plastix, 2022-12-16:
My X1c & AMS upgraded firmware today (1.03.00.00 & 00.00.05.75), and this problem is completely resolved.

I printed Vase mode with and without timelapse, and no prints had the pause or visible anomaly now.
I'm closing this out as a mismatch of New Studio with Old Firmware.

closed 2022-12-16 by Max-Plastix