bambulab/BambuStudio #847: Power Consumption of X1/X1C when idle

Manfredo1952, 2022-12-16:
**Is your feature request related to a problem?  

When the printer is idle, power consumption is about 25 to 30 W. 
This can be reduced to 10 what if at the end of a print  and the stepper motors are disabled. 
 
These days when energy cost is quite high, it would be quite nice to save some energy and money. 
Machine EndCode ends with "M17 X0.8 Y0.8 Z0.5 ; lower motor current to 45% power"

**Describe the solution you'd like**
Instead of 45% power for stepper motors, this should be Zero
This can be achieved by using " M18 " instead

Power consumption after M18 and machine idle  reduces to 10 Watt

Thanks
[image]

SaltWei, 2022-12-16:
Machine gcode is changed by printer team and this is printer issue. Mark with Printer label.
We will transfer this ticket to them.

Manfredo1952, 2022-12-17:
I have noted that the new FW of the printer will now idle down to lower consumlotion after 20 minutes. But why 20 minutes ?  Suggest to idle down within 2-3 minutes, since it does not make sense to consume energy. Thanks

SaltWei, 2022-12-17:
Sorry,this is github issue collection about Studio and we don't know. We will transfer this message to printer team again.
For printer issue, user  can move to https://forum.bambulab.com/ for better and more efficient support. Thanks!

SaltWei, 2022-12-17:
OK. New message from printer team about why it's 20min.
The reason is keeping position for a longer time in case of pop-up window to remind user to do zero returning when user moves printer axis. 2-3min is too short.

closed 2022-12-17 by Manfredo1952