bambulab/BambuStudio #849: [Issue] Dark Mode Inverts Color of Black and White Filaments Under AMS In Device Tab

hifihedgehog, 2022-12-16:
This issue occurs In the *Device* tab when dark mode is enabled. Under the AMS filament selection area, if you have either a pure white or black color selected for a filament, the color block preview (by clicking on the pencil edit icon ✏️) will show up inverted in color, meaning black will be white and white will be plate. Specifically, it is the bounded box to the right of *Color* that shows a preview of the color that is showing up as black instead of white (or white instead of black) when in dark mode. As far as I can tell, only black and white is impacted. In the example below, you can see that black is shown yet it is a white filament that is selected. I see the same behavior with the black filament in a different slot as well. Turning off dark mode fixes this behavior.

[image]
[image]

bambu123, 2022-12-16:
@SaltWei Please follow this picture to check if system textColor is modified.
[image]

hifihedgehog, 2022-12-16:
> @hifihedgehog Please follow this picture to check if system textColor is modified.
> [image]

I am on **Windows** and not macOS (note the window buttons in the upper righthand corner of the screen). This is what I see here. Color is correct in the Prepare tab. **However**, this is not the problem area. The issue is in the *Device* tab where I had noted earlier.

[image]

closed 2023-03-02 by SaltWei