bambulab/BambuStudio #851: BS 1.4.0.17 Failed to Get Network Plugin

alexsanchiz, 2022-12-16:
Everything was working OK untill I updated to version 1.4.0.17. 
Bambu Studio can't get network plugin and I can't access to my printer from BS now.
Nothing has changed on my PC, BS was working well. Using W7. 
Tried to do a clean install, deleted user data, etc. Nothing seems to work.

alexsanchiz, 2022-12-16:
Fixed installing BS on another PC and copying C:\Users\%USERPROFILE%\AppData\Roaming\BambuStudio\plugins archives from there to my PC.
bambu_networking and BambuSource

lanewei120, 2022-12-16:
the network plugins have been updated and stored on new servers with cdn
there is a known issue that windows 7/8 can not download it by using SChannel in curl

we are trying to switch to use openssl instead
and if it is doable, we will release a hotfix version.

currently you can download if from https://public-cdn.bambulab.com/upgrade/studio/plugins/01.04.00.15/win_01.04.00.15.zip
and extract them into C:\Users\user.name\AppData\Roaming\BambuStudio\plugins manually

sorry for this inconvinience

alexsanchiz, 2022-12-16:
> the network plugins have been updated and stored on new servers with cdn there is a known issue that windows 7/8 can not download it by using SChannel in curl
>
> we are trying to switch to use openssl [...]

Hi, 
Copyed that and just works, but now happens the same with the virtual camera tools

lanewei120, 2022-12-16:
you can download it from https://public-cdn.bambulab.com/upgrade/studio/cameratools/01.04.00.01/win_01.04.00.01.zip
currently, and extract into C:\Users\user.name\AppData\Roaming\BambuStudio\cameratools

lanewei120, 2023-03-01:
already in v1.5 just released

closed 2023-03-01 by lanewei120