bambulab/BambuStudio #852: [*Issue*] BS 1.4.0.17 Failed to Get Network Plugin

BambuLabDiscord, 2022-12-16:
Everything was working OK untill I updated to version 1.4.0.17.
Bambu Studio can't get network plugin and I can't access to my printer from BS now.
Nothing has changed on my PC, BS was working well yesterday. Using W7.
Tried to do a clean install, deleted user data, disabling antivirus, firewalls, etc. Nothing seems to work.

lanewei120, 2022-12-16:
the network plugins have been updated to new servers with cdn
there is a known issue that windows 7/8 can not download it by using SChannel in curl

we are trying to switch to use openssl instead
and if it is doable, we will release a hotfix version.

currently you can download if from https://public-cdn.bambulab.com/upgrade/studio/plugins/01.04.00.15/win_01.04.00.15.zip
and extract them into C:\Users\user.name\AppData\Roaming\BambuStudio\plugins manually

sorry for this inconvinience

lanewei120, 2023-01-31:
already fixed in v1.4.2

closed 2023-01-31 by lanewei120

bjornsyse, 2023-03-11:
I have this problem now again from 1.05. Where can I find the plugin manually? 

I cannot access my printer anymore.,

lanewei120, 2023-03-13:
> I have this problem now again from 1.05. Where can I find the plugin manually?
> 
> I cannot access my printer anymore.,

do you mean it worked in previous version?
what is your OS using?
and could you send the log in C:\Users\user.name\AppData\Roaming\BambuStudio\log for us to check

it maybe a different issue, most likely network related.
anyway, you can download it from 
https://public-cdn.bambulab.com/upgrade/studio/plugins/01.05.00.10/win_01.05.00.10.zip for windows

bjornsyse, 2023-03-13:
I'm on windows 10. I found the file manually in the log file but even after placing the file in plugins folder I still can't get around the error msg when launch the software. 

Log file  attached

bjornsyse, 2023-03-13:
Are there any domains changed for cdn's since last time? I've had to whitelist them with my proxy provider the last time.

lanewei120, 2023-03-13:
> 

still can not find the log files
it should be something wrong...

bjornsyse, 2023-03-13:
[debug_Mon_Mar_13_08_23_44_10160.log.zip](https://github.com/bambulab/BambuStudio/files/10954841/debug_Mon_Mar_13_08_23_44_10160.log.zip)
Sorry, log file was refued because of file extension, I zipped it now.

lanewei120, 2023-03-13:
> [debug_Mon_Mar_13_08_23_44_10160.log.zip](https://github.com/bambulab/BambuStudio/files/10954841/debug_Mon_Mar_13_08_23_44_10160.log.zip) Sorry, log file was refused because of file extension, I zipped it now.

2023-03-13 08:23:49.368140[Thread 0x000096f4]:[install_plugin]Archive read exception:boost::filesystem::remove: Access is denied [system:5]: "C:\Users\tmgxbs\AppData\Roaming\BambuStudio\plugins/BambuSource.dll"

from the log, it seems the dll is occupied by someone else
are you using multiple BambuStudio at the same time?
if yes, please exit from other BambuStudio before install the plugins.

Also you can check in the task manager to check whether there is another instance there, if yes, please kill that ones firstly

bjornsyse, 2023-03-13:
No other instances running no. I double checked in the Task manager.

bjornsyse, 2023-03-14:
I launched as administrator now, and the plugin installed. But when I launch as regular user after that, I still get an error it can not be installed.

lanewei120, 2023-03-14:
could you give a snapshow of the directory "C:\Users\tmgxbs\AppData\Roaming\BambuStudio\plugins"
and you can attach the log this time here.

bjornsyse, 2023-03-14:
Reporting that after another restart of the computer, I Can now start the program as regular user too, and the installation happened again in that session. 

Folder you mentioned has these files: 

`backup\
bambu_networking.dll
BambuSource.dll`

lanewei120, 2023-03-14:
> Reporting that after another restart of the computer, I Can now start the program as regular user too, and the installation happened again in that session.
> 
> Folder you mentioned has these files:
> 
> `backup\` `bambu_networking.dll` `BambuSource.dll`

so could you upload the log here this time?

bjornsyse, 2023-03-14:
Sorry, here comes. There were multiple but these are all from today. 
[log.zip](https://github.com/bambulab/BambuStudio/files/10965553/log.zip)

lanewei120, 2023-03-14:
from log, it seems the plugins worked well

so it seems the application can work as normal now

do you have any other issues?

bjornsyse, 2023-03-14:
Hi, no thank you for your troubleshooting - now it works. I'm behind so many firewalls/virus programs/endpoint protections and proxy's you'd never know where it might have failed. There is nothing I can do to turn them off unfortunately.

bjornsyse, 2023-05-08:
So, had the same problem today when trying to install the 1.6.2.4. 

What finally worked, was:

1. after installation - start Bambu studio as administrator. Network plugin get's installed automatically. 
2. Copy `bambu_networking.dll` from `c:\Users\[MYADMINACCOUNT]\AppData\Roaming\BambuStudio\plugins\` to `c:\Users\[MYUSERACCOUNT]\AppData\Roaming\BambuStudio\plugins\`
3. Start bambu studio as regular user.

lanewei120, 2023-05-09:
Glad to hear it, thanks for the information~

lanewei120, 2024-11-29:
@Denislaaav please report a new issue, our QA will contact you there