bambulab/BambuStudio #853: Error: Stopped [-20000]! when trying to play video stream from printer

Wixely, 2022-12-16:
**Describe the bug**
When trying to play the feed on the Device page, I get an error Stopped [-20000]!

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Devices tab
2. Click on Play icon below the stream
3. Stream does not load and error appears in status bar below the video "Stopped [-20000]!"
4. It will loop, Playing, Stopped, Playing, Stopped, etc.

**Expected behavior**
The video should play

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro, 21H2 19044.2251
 - Bambu Studio 1.4.0.17
 - Printer: Bambu Lab X1 Carbon - fw: 01.01.01.00

This is reproducable on multiple devices (desktop + laptop).
Both devices have windows media player installed.
Clicking the error brings me to a wiki page but the error doesn't exist on the page: https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view
The printer is NOT in lan mode
Issue was not present in previous version of Bambu Studio.
Printer and computers restarted multiple times.
Network does not appear to be an issue - printer still prints, timelapse videos still record and are viewable once finished.
Any help appreciated, thanks!

Logs:

```
2022-12-16 11:49:00.146019[Thread 0x00004654]:MediaPlayCtrl::SetStatus: Initializing...
2022-12-16 11:49:00.836698[Thread 0x00007184]:MediaPlayCtrl camera_url: bambu:///B2YHESFN965CTS8F111A?authkey=<redacted>&passwd=<redacted>&region=us, machine: 00M00A261900648
2022-12-16 11:49:00.837524[Thread 0x00004654]:MediaPlayCtrl::SetStatus: Loading...
2022-12-16 11:49:00.838531[Thread 0x00003fc4]:MediaPlayCtrl: start load
2022-12-16 11:49:00.838531[Thread 0x00003fc4]:wxMediaCtrl2::Load: clsid {233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA} path C:\Users\X13\AppData\Roaming\BambuStudio\plugins\BambuSource.dll
2022-12-16 11:49:00.840531[Thread 0x00003fc4]:MediaPlayCtrl: end load
2022-12-16 11:49:00.850295[Thread 0x00004654]:MediaPlayCtrl::onStateChanged: -1, last_state: 5
2022-12-16 11:49:01.085713[Thread 0x00003208]:process_task{ type:Backup, id:0, path:, object:0, extra:343}
2022-12-16 11:49:01.085713[Thread 0x00004654]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:343}
2022-12-16 11:49:01.221530[Thread 0x00004654]:[BambuTunnelTutk::open] IOTC_Connect_ByUIDEx(B2YHESFN965CTS8F111A).......
2022-12-16 11:49:01.945080[Thread 0x00004654]:[BambuTunnelTutk::open] Device[B2YHESFN965CTS8F111A] is from [192.168.0.241:37104] Mode[LAN] NatType[2] IOTCVersion[3040403]
2022-12-16 11:49:01.945080[Thread 0x00004654]:[BambuTunnelTutk::open] Device is from 192.168.0.241:37104[B2YHESFN965CTS8F111A] Mode=LAN
2022-12-16 11:49:01.945080[Thread 0x00004654]:[BambuTunnelTutk::open] Device info VER[3040403] NAT[2]
2022-12-16 11:49:02.006347[Thread 0x00004654]:[BambuTunnelTutk::open] avClientStartEx success avIndex[0] resend[1] two_way_streaming[1]
2022-12-16 11:49:02.203038[Thread 0x00004654]:[BambuTunnel::read_sample] would block [1]
2022-12-16 11:49:02.285101[Thread 0x00004654]:[BambuTunnel::read_sample] would block [2]
2022-12-16 11:49:02.342324[Thread 0x00004654]:[BambuTunnel::read_sample] would block [3]
2022-12-16 11:49:02.443214[Thread 0x00004654]:[BambuTunnel::read_sample] would block [4]
2022-12-16 11:49:02.544035[Thread 0x00004654]:[BambuTunnel::read_sample] would block [5]
2022-12-16 11:49:02.644067[Thread 0x00004654]:BambuTunnel::start_stream ok
2022-12-16 11:49:03.140349[Thread 0x00004654]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2022-12-16 11:49:03.140349[Thread 0x00004654]:MediaPlayCtrl::onStateChanged: size: 1168x720
2022-12-16 11:49:03.140349[Thread 0x00004654]:MediaPlayCtrl::SetStatus: Playing...
2022-12-16 11:49:03.567905[Thread 0x00004654]:MediaPlayCtrl::onStateChanged: 2, last_state: 0
2022-12-16 11:49:03.567905[Thread 0x00004654]:MediaPlayCtrl::onStateChanged: 2, last_state: 2
2022-12-16 11:49:03.567905[Thread 0x00004654]:[BambuTunnel::read_sample] error [-20000]
2022-12-16 11:49:03.638700[Thread 0x00004654]:[BambuTunnelTutk::close] avClientStop calling....
2022-12-16 11:49:03.638700[Thread 0x00004654]:[BambuTunnelTutk::close] avClientStop OK
2022-12-16 11:49:03.638700[Thread 0x00004654]:[BambuTunnelTutk::close
```

theogott, 2022-12-17:
I have that too. I have checked  that it should not be a Firewall Problem.
Also the picture came shortly and then disappeared. 
I got this error "Stopped[-20000]!". Now it says "Laufend..." (german means "Running...".
But no picture. 
This has been working fine with all other versions.

PS: Besides that the new version looks amazing!

[image]

Wixely, 2022-12-17:
New firmware has become available since I posted this, but the bug remains. (fw: 01.03.00.00)

Maybe 3 times in total, if I leave it looping I will eventually get a video stream. Once I have the stream it appears to work fine, but navigating away from the Devices page puts be back to step one.

It's working fine on the mobile app, and if I follow the steps to stream via OBS it works every time without issue in OBS.

jjumbc, 2022-12-19:
no longer working for me on latest update either. Printer on latest firmware as of this posting. X1 non-carbon with camera installed

FlowForm, 2022-12-19:
Getting the same error on my X1C.

swamp-21, 2022-12-19:
I'm having the exact same issue with the latest version of BS. I've reinstalled, logged out and back in, updated the firmware on the printer, power cycled the printer, etc. No issues on the Handy app. Uninstalling the newest version of BS and reinstalling the previous version (v01.03.01.01) corrects the issue with the camera.

Running Windows 10 Pro

theogott, 2022-12-20:
Jf anybosy @Bambu Labs needs help with Debugging or wants to have some Beta tested, contact me.
I can help.

ppeccard, 2022-12-20:
Hi, I have the same issue too, any help or suggestion will be appreciated.

ppeccard, 2022-12-20:
I can confirm that a older version like the 1.03.01.01 works fine with the video

bambu123, 2022-12-21:
Let's start with collecting more logs. With this DLL in the zip file, I add more logs. Please replace the one in the directory "%APPDATA%\BambuStudio\plugins".
Run and send me the debug log.
[BambuSource.zip](https://github.com/bambulab/BambuStudio/files/10273684/BambuSource.zip)

theogott, 2022-12-21:
Before i install a strange DLL i need to know from who it is.
Also the DLL is not digitally signed.

bambu123, 2022-12-21:
It is one module of the Bambu network plugin, that works for the video stream.
This one in the zip is the one with more debug logs. So I just provide here instead update from plugins updating.
After debugging, you can safely delete it and let the plugins update to download the public ones.

theogott, 2022-12-21:
Thats not the question. I just need that you email it to me with your contact data so I know who you are, are you from Bambu Labs or a private contributor ... or you can just sign it digitally with the Bambu Signature.
This a question of safety.

bambu123, 2022-12-21:
@theogott email sent, please check.

theogott, 2022-12-21:
Here is my E_Mail adress, i am also available in Skype.
If you can not officially sign the DLL, then i need to know who you are, so please include contact data.
[image]

theogott, 2022-12-21:
I have sent you something to your E-Mail please answer an then we can start.

Standroyd, 2022-12-21:
I've been having this same issue since the recent update.
I noticed LAV Filters video codec was being loaded when the video started, so I tried manually updating those to the latest version (https://github.com/Nevcairiel/LAVFilters/releases) - but still no luck.
After uninstalling LAVFilters, the video started working everytime - no more -20000 error. The video quality appears poorer though.
I reinstalled LAVFilters again and started getting the -20000 error again - uninstalled and the problem went away again.
Hope this info is useful.

bambu123, 2022-12-21:
Hi, Theo: Thanks for your help. For the -20000 problem, we want to collect more logs. Attached is the video playback component with logs added. （newer that the github ones) How To Do:Copy BambuSource.

theogott, 2022-12-21:
Could you install the DLL from Chenmao?
I do not even have that Folder plugins"", and in the folder "plugin" there is no such DLL.

theogott, 2022-12-21:
I have sent you all relevant Logs to your e-mail.
Also i found that plugins folder.
Please check your mail and tell me if i can help in other ways. 
You can also reach me on Skype.

theogott, 2022-12-21:
@Chunmao, I have seen that even the original DLL's are not digitally signed. 
As a professional Company that sells products, all your executables and DLL's should be digitally signed.
A digital Signature prevents files from being corrupted.
The easiest way to do so is, using "[Centurion Setup](https://secure.2checkout.com/affiliate.php?ACCOUNT=GAMMADYN&AFFILIATE=140120&PATH=http%3A%2F%2Fwww.gammadyne.com%2Fcenturionsetup.exe)".
It will automatically digitally sign all your Files while the Setup is created.  
Its possibly the best program for that task and very cheap (120$ or such).

bambu123, 2022-12-21:
Hi, Gittwald: Please help check if this one works. Talking about digitally signing, yes, it should be signed. I have fed back to the relevant personnel.

theogott, 2022-12-21:
What exactly should I check? Send it to me with mail or post it here please.

bambu123, 2022-12-21:
Send attachments again.

theogott, 2022-12-21:
[image]

theogott, 2022-12-21:
Maybe send it with Mail, Wetransfer to my mail?

theogott, 2022-12-21:
Great, you did it again, ChunMao. Work, you can ship it!

[image]
[BambuSource.zip](https://github.com/bambulab/BambuStudio/files/10277069/BambuSource.zip)

bambu123, 2022-12-21:
OK, I will update the plugin as soon as possible. Thank you for your help again!

theogott, 2022-12-21:
You are welcome. If there is anything else to test you can contact me anytime.

bambu123, 2022-12-21:
Please play the video for a while to see if it is all ok.

theogott, 2022-12-21:
Its playing .... I keep you updated if there are further issues.

[image]

bambu123, 2022-12-21:
About 5 minutes is enough.

theogott, 2022-12-21:
It still running .... 
[image]

theogott, 2022-12-21:
Please note that 
https://github.com/bambulab/BambuStudio/issues/870
ca also be closed with this together.

bambu123, 2022-12-21:
yes,  after the new plugin was published.

ppeccard, 2022-12-21:
Thanks Theogott and bambu123 it works also for me, great job

bambu123, 2022-12-21:
We have published the network plugin with version 01.04.00.16.
Restart Bambu studio to update the plugin.
It is a silent update, the first restart will only download the plugin, and the secondary will update it.

theogott, 2022-12-21:
This means that the Bambu Studio contains code that can silently update Files on my computer.
This will increase the value of your company in USA greatly. 
I suggest that you only update files after asking the user to do so.

bambu123, 2022-12-21:
A good suggestion too.

bambu123, 2022-12-21:
We'll add a tip for the plugin-updated in the next RC.

theogott, 2022-12-21:
Yes, just think about i give you a program and it will "Update silently" stuff on your computer.
You will never use this sort of program in critical infrastructure.

Not because of i may trust you or not, 
but also maybe the mechanism can be hacked and someone else can use it.
So if you have an update be polite and ask the user if its ok.

SaltWei, 2023-03-02:
> When trying to play the feed on the Device page, I get an error Stopped [-20000]!

Should be fixed in v1.5.0 public beta

closed 2023-03-02 by SaltWei