bambulab/BambuStudio #854: Support overhang slow down and arachne at the same time.

SaltWei, 2022-12-16:
See discussing in https://github.com/bambulab/BambuStudio/issues/466

psiberfunk, 2022-12-16:
Thanks for forking this off the parent task !

finsdotsurf, 2023-01-31:
I asked chatgpt and it came up with some ideas that could help solve this problem:

Pre-calculate and cache the serial of width values and expolygons based on a set of common line widths that are likely to be used in practice. This would allow you to avoid recalculating the serials and expolygons every time the line width is changed.

Another approach is to use approximation techniques to reduce the number of segments and width values that need to be calculated. For example, you could use a piecewise linear approximation to represent the relationship between the overhang degree and line width, which would allow you to perform the calculations more quickly by using fewer segments and width values.

Some potential approximation techniques:

Use a piecewise linear approximation to represent the relationship between the overhang degree and line width, allowing you to perform the calculations more quickly by using fewer segments and width values.

B-spline Approximation, a form of curve fitting that uses a series of polynomial functions to help reduce the number of calculations required and to achieve a more accurate representation of the relationship between the overhang degree and line width.

Simplification algorithms, such as the Ramer-Douglas-Peucker algorithm, to reduce the number of segments in the line.

Interpolation methods, such as linear or polynomial interpolation, to estimate the overhang degree and line width based on a smaller set of reference points.

QingZhangBambu, 2023-04-21:
We've supported that feature, and you can get that in version 1.6

closed 2023-04-21 by QingZhangBambu