bambulab/BambuStudio #855: BS 1.4.0.17 won't slice

Benderrodrigu3z, 2022-12-16:
New BS Slicer won't finish slicing objects when the old version still does, the program hangs here and freezes out, if you click the stop button it freezes it out sooner

[image]

lanewei120, 2022-12-20:
hello

is it a 100% reproducable issue or not?
could you upload the 3mf to us for debugging?

lanewei120, 2022-12-31:
@Benderrodrigu3z 

hello

is this issue still exist there?
if yes
could you help to upload the 3mf for us to have a debug

Benderrodrigu3z, 2022-12-31:
Still an issue, I don't have that one but here is another one that does the same thing
[Poop Bucket.3mf.zip](https://github.com/bambulab/BambuStudio/files/10327791/Poop.Bucket.3mf.zip)

SaltWei, 2023-03-02:
Still exist on v1.5.0 public beta and 100% to be reproduced.
@tangketan  Please have a look

tangketan, 2023-08-10:
There are too many tree nodes and causing the generation very slow. Please use the "tree hybrid" style to workaround for now.