bambulab/BambuStudio #857: Stream not 1080p when set to 1080p in Bambu Studio

VACIndustries, 2022-12-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Followed these instructions here https://wiki.bambulab.com/en/software/bambu-studio/virtual-camera
2. Selected 1080p in Bambu Studio Resolution under "Go Live"

**Expected behavior**
Stream in 1080p

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1.4

Benderrodrigu3z, 2022-12-16:
you might have to also hit 1080p on the printers Screen, its under Tab 4 General

VACIndustries, 2022-12-17:
Tried this...tried updating the firmware on the printer and restarting the printer and verifying the 1080p is selected on the printer's display as well -- still 720p resolution

bambu123, 2023-01-05:
You have to restart the stream after switching the frame size of the printer.
OBS has also to be restarted because it may not respond to in-stream frame size changes.
Also, OBS is not telling the actual size of the stream.  You can use Video Lan to open the SDP file to see the size of the stream.
The size is really 1760 * 1080, not 1920 * 1080 for not showing outside of the printer.

bambu123, 2023-01-10:
@VACIndustries Has the problem been solved?

VACIndustries, 2023-01-10:
I think so. Will reopen if I find otherwise. Thanks for the recommendations

closed 2023-01-10 by VACIndustries

bambu123, 2023-01-11:
Also, update OBS to 29.0.0 will respond to frame size changes.