bambulab/BambuStudio #860: Increasing Height Does NOT work when Cutting at an angle

Benderrodrigu3z, 2022-12-16:
If I cut Rotate the Cut the Height Option becomes Useless

[image]

tangketan, 2023-01-11:
It's not a bug. By definition, "height" controls the perpendicular distance between the cut plane and the z=0 plane. If the two planes are already perpendicular, then the concept of "height" becomes meaningless. Of course you can't control the "height" anymore. I can disable the input box in such case, if that's what you like.

Benderrodrigu3z, 2023-01-11:
Then why is there an UP ARROW? Therefore it is a height, if your unwilling to give a necessary control like this, at least give a left to right control when the cut is rotatedSincerely,Shane FSent from my iPhone13

tangketan, 2023-01-11:
@Benderrodrigu3z the arrows are for adjusting the rotation angles of the cut plane, not the height. If you want to move the plane, you need to grab the yellow cube and then move your mouse.
By the way, I've disabled the height setting when the cut plane is not completely horizontal in the next version, hoping it'll be less confusing.

Benderrodrigu3z, 2023-01-11:
I just told you to not do that as it will cause more confusion for keep top half or keep bottom half, now there is no way to tell, how did you miss that sarcasm?Sincerely,Shane FSent from my iPhone13

tangketan, 2023-03-02:
"Height" is officially disabled when cut plane is not horizontal. Issue closed.

closed 2023-03-02 by tangketan