bambulab/BambuStudio #862: Cut Breaking Dimensions

Benderrodrigu3z, 2022-12-16:
Cut Breaking its own Dimensions, exiting out of cut then back in for now corrects the bug

Why?
[image]

Good
[image]

zhimin-zeng-bambulab, 2023-01-04:
@Benderrodrigu3z  Could you describle your problem in detail, and provide the steps to reproduce the problem? It will help us to solve the problem.

zhimin-zeng-bambulab, 2023-01-04:
The phenomenon I see is that when exiting out of cut then back in, the cutting surface will be reset to the initial position.

Benderrodrigu3z, 2023-01-04:
its random is the issue

Benderrodrigu3z, 2023-01-04:
and is there a way to make the cutting plane white area view larger? to cut a corner I have to rotate it 8 times to even get the white area to touch the print, half the time I rotate the wrong way and its not touching the cutter that I can tell because the plane isnt in view of the object, purhaps it should color the print a slightly off shade as it hits the plane so that rotating it is no longer an issue, pics to come

Benderrodrigu3z, 2023-01-04:
This should not be the same color for both sides, make one side a darker side and one side a lighter side incase the filament color is white
[image]

Benderrodrigu3z, 2023-01-04:
also Edit undo should take it back to the last rotation for the cutting field, that doesn't work, clicking out kills the plane when your still in plane cut view.

zhimin-zeng-bambulab, 2023-02-09:
We plan to improve the cutting feature in the future, it will take some time.

zhimin-zeng-bambulab, 2023-04-18:
we have improved the cutting feature, please have a try.

zhimin-zeng-bambulab, 2023-08-18:
Does this problem still exist in the latest version?

zhimin-zeng-bambulab, 2023-10-11:
Close as no response

closed 2023-10-11 by zhimin-zeng-bambulab