bambulab/BambuStudio #863: Delete an Object while still slicing doesn't Cancel Slicing

Benderrodrigu3z, 2022-12-16:
We Can now delete an object while Slicing, this is occurring on Mac in this new version which breaks and freezes the BS Slicer because it doesn't stop the slicing process and ends in a crash, sometimes you can save it by clicking stop but that usually takes another 20sec for it to kick back

[image]

lanewei120, 2022-12-20:
for mac, there will a crash report if crashed
could you send the crash report to us?

also is it a 100% reproducable issue or not?
and could you help to upload the 3mf here?

lanewei120, 2023-04-08:
we tried it on the newest version and can not reproduce it

as this is a issue long long again
we close it currently

if still has this issue, please reopen it and provide the crash report

thanks

closed 2023-04-08 by lanewei120