bambulab/BambuStudio #864: Option to select an object as target for filament purge (use it to print something instead of throwing it away)

c2h5oh, 2022-12-16:
**Is your feature request related to a problem? Please describe.**
A lot of filament is wasted on multicolor prints because it gets purged and thrown away on each color change. It could be used to print something where we don't care about color(s) instead.

**Describe the solution you'd like**
When printing multicolor object I should be able to select another object on the same plate that will be printed using filament that would otherwise be purged when switching colors.

**Additional context**
I often print things that are intended as tools and I don't care how they look. I'd be happy to print them in part with filament that would otherwise be wasted during a multicolor print, even if that meant weird color transitions.

rncfiles, 2022-12-16:
I think this already exists.  
1) Add a second object
2) select it
3) under "other" select "flush into this object" option

c2h5oh, 2022-12-17:
You are right - it does exactly what I described.

closed 2022-12-17 by c2h5oh