bambulab/BambuStudio #865: Make our own gradient/rainbow prints by switching filaments without purging

c2h5oh, 2022-12-16:
**Is your feature request related to a problem? Please describe.**
1. Almost all gradient/rainbow filaments only work with larger prints, because they change color every 15-20 meters or every 50-65 grams. This doesn't work for small prints and even medium-large vase mode prints - you don't even get a single full transition on the entire print.
2. For the same reason as above you can't print larger prints with a quick but smooth color transition - filament has full control over that how colors change.
3. Gradient/rainbow filaments are harder to buy and the few that change color every 5-8m are impossible to buy in EU.

**Describe the solution you'd like**
I'd like to be able to select a filament change, without purge:
- every x layers, or
- every x meters of filament used, or
- every x full transitions (e.g. if your best guess if full color transition required n meters of filament after switch print x*n meters before switch)
- once filament on spool runs out (great to use up last bits of filament or free samples you often get with your purchases)

Proposed UI:
Add first filament from AMS + select condition switch to the next one
Add second filament + select condition to switch to next one
...
Add last filament + select optional condition to switch to the first one (if not set it will finish printing with this filament, if set it will loop back to 1st one and repeat)

I'd like to also be able to see a rough visualization of how it's going to look - most likely after slicing, because it will have to account for supports, fill rates etc.

**Describe alternatives you've considered**
Setting purge volume to 0 and manually painting objects. It kind of works, but is a major pain and is basically guesswork, because you have no visualization of how it's going to look like until you print it.

**Additional context**
This is a great example of something that would not be possible without AMS. Sure, a multi nozzle printer can switch between colors faster and without purging, but it can't blend colors.

rncfiles, 2022-12-20:
This is good idea and well thought out.

SaltWei, 2023-03-02:
[image]
The nozzle volume is small and the mixed line is too short after switching filament without purging.
This is a good idea, but we evaluate that maybe the effect is unsatisfactory.