bambulab/BambuStudio #871: Mac > V.1.4.0.17 - App extremely laggy in Seam Painting & Color Painting

UncleJessy, 2022-12-17:
In the latest bambu studio v1.4.0.17 on Mac OS

1. Import a model
2. Add more than one Material
3. Open Seam Painting - Try to Navigate around the object
4. Open Color Painting - Try to navigate around the object
5. The app will respond poorly/slowly 

This was not an issue in the previous release and is extremely laggy to use now

YifanWuBambu, 2022-12-18:
Hi @UncleJessy ,

I just tried on M1 and M2 MacBook, but did not reproduce the issue.
If possible, could you share your 3mf (zip to a .zip file) file with us?

Thanks a lot!

destinybonavita, 2022-12-18:
> Hi @UncleJessy ,
> 
> I just tried on M1 and M2 MacBook, but did not reproduce the issue. If possible, could you share your 3mf (zip to a .zip file) file with us?
> 
> Thanks a lot!

We're experiencing this issue as well on a 2.6 GHz 6-Core Intel Core i7 MacBook Pro. 
We're also running macOS 13.0 (22A380). 
It's not project specific either, any new project is insanely slow with painting even with models of 30k triangles that didn't have issues with older versions of BambuStudio.

UncleJessy, 2022-12-20:
M1 Max Macbook Pro 32gb of RAM MacOS 12.1 

This happens with a basic benchy or virtually any model of size when I try to use any paint function. I'm also noticing the CPU spikes over 100% as soon as I try these functions. 

I will upload a 3mf & video shortly

UncleJessy, 2022-12-20:
> We're experiencing this issue as well on a 2.6 GHz 6-Core Intel Core i7 MacBook Pro. We're also running macOS 13.0 (22A380). It's not project specific either, any new project is insanely slow with [...]

I also posted about this on twitter and there are also others mentioning the same issue

YifanWuBambu, 2022-12-20:
> I also posted about this on twitter and there are also others mentioning the same issue

Thanks a lot for the information.

YifanWuBambu, 2022-12-20:
> We're experiencing this issue as well on a 2.6 GHz 6-Core Intel Core i7 MacBook Pro. We're also running macOS 13.0 (22A380). It's not project specific either, any new project is insanely slow with [...]

I tried on an i7 Mac Mini, but still not reproduce this issue.
We will keep reproducing it. It is our high priority task.

UncleJessy, 2022-12-20:
> I tried on an i7 Mac Mini, but still not reproduce this issue. We will keep reproducing it. It is our high priority task.

Here is a sample video of the issue in action. 
https://www.youtube.com/watch?v=jPPgEho-0Bg

Here is the 3MF file in ZIP
https://drive.google.com/file/d/115CHtpIj7Axv7hEGfYjuZplrN0mjUZuz/view?usp=sharing

doomcrewinc, 2022-12-20:
I cannot reproduce this on my M1 Macbook Pro, however on an i7 2.6ghz Macbook Pro (12.6) I have experienced this using every model from a simple Benchy, to a 2 million+ triangle model, and the results are the same.  Unbearable lag.  

Switching from paint to a different mode and the lag disappears, switch back and it starts lagging again making it impossible to paint.

lanewei120, 2022-12-21:
> Here is a sample video of the issue in action. https://www.youtube.com/watch?v=jPPgEho-0Bg
>
> Here is the 3MF file in ZIP [...]

thanks very much for this video

is the lagging only happened when doing rotation?
if doing scale(by mouse wheel),  is it lagging too

UncleJessy, 2022-12-21:
> > > > > Hi @UncleJessy ,
> > > > > I just tried on M1 and M2 MacBook, but did not reproduce the issue. If possible, could you share your 3mf (zip to a .zip file) file with us?
> > > > > Thanks a lot!
> > > > 
> > > > 
> > > > We're experiencing this issue as well on a 2.6 GHz 6-Core Intel Core i7 MacBook Pro. We're also running macOS 13.0 (22A380). It's not project specific either, any new project is insanely slow with painting even with models of 30k triangles that didn't have issues with older versions of BambuStudio.
> > > 
> > > 
> > > I tried on an i7 Mac Mini, but still not reproduce this issue. We will keep reproducing it. It is our high priority task.
> > 
> > 
> > Here is a sample video of the issue in action. https://www.youtube.com/watch?v=jPPgEho-0Bg
> > Here is the 3MF file in ZIP https://drive.google.com/file/d/115CHtpIj7Axv7hEGfYjuZplrN0mjUZuz/view?usp=sharing
> 
> @UncleJessy thanks very much for this video
> 
> @UncleJessy @doomcrewinc @destinybonavita is the lagging only happened when doing rotation? if doing scale(by mouse wheel), is it lagging too or is it happened in every operations under paint mode
> 
> I also built two images which may fix this issue, however as we can not reproduce it here could you help to have a check on your machine which has this issue? [debug version 1.4.0.19](https://www.dropbox.com/s/n3ph8qdibtmvg1x/Bambu_Studio_mac-v01.04.00.19-20221221204643.dmg?dl=0) [debug version 1.4.0.20](https://www.dropbox.com/s/d50md47xh4afpnf/Bambu_Studio_mac-v01.04.00.20-20221221213610.dmg?dl=0)

Yes, it happens with any movement/function that you try to perform while using ANY of the paint like functions
Support Painting
Seam Painting
Multi Color Painting etc
Rotate/zoom/Clicking / Dragging/ making selections its all extremely laggy and the CPU spikes 

I will download and try those two out
thanks

UncleJessy, 2022-12-21:
> @UncleJessy thanks very much for this video
>
> @UncleJessy @doomcrewinc @destinybonavita is the lagging only happened when doing rotation? if doing scale(by mouse wheel), is it lagging too or is it [...]

THAT FIXED THE ISSUE! 

I tried both builds starting with .19 > it worked properly!
I then installed .20 > it worked properly!

I can now rotate/zoom/pan/click/drag etc in Paint supports, seams, multi color and it now works again properly!

lanewei120, 2022-12-22:
> THAT FIXED THE ISSUE!
>
> I tried both builds starting with .19 > it worked properly! I then installed .20 > it worked properly!
>
> I can now rotate/zoom/pan/click/drag etc in Paint supports, seams, [...]

@UncleJessy 
thank you so much
could you also help to upload a log file in /Users/user.name/Library/Application Support/BambuStudio/log/debug_xxx.log

then I can double confirm the detailed root cause.

UncleJessy, 2022-12-27:
> @UncleJessy thank you so much could you also help to upload a log file in /Users/user.name/Library/Application Support/BambuStudio/log/debug_xxx.log
>
> then I can double confirm the detailed root cause.

This also appears to be resolved with the v1.4.1 beta

YifanWuBambu, 2022-12-27:
Yes. It was caused by the wire frame feature introduced in v1.4. We removed it in v1.4.1 to not block using the painting features.

SaltWei, 2023-08-18:
This seems to be solved old issue. Close

closed 2023-08-18 by SaltWei