bambulab/BambuStudio #872: Get estimation data via CLI

sguter90, 2022-12-17:
I want to get estimation data of a print via CLI to automate stuff (e.g. get print time for all files in a directory or building a web interface)

**Is your feature request related to a problem? Please describe.**
https://github.com/bambulab/BambuStudio/issues/76 introduced price estimation

**Describe the solution you'd like**
I would like to use the CLI to get filiament (m/g), cost, model printing time and total time as displayed in the preview tab. (maybe more if it makes sense)
With the PrusaSlicer I do something like this (on MacOS):
```
/Applications/Original\ Prusa\ Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer --export-gcode --layer-height 0.4 --output example.gcode ./example.stl
```

This command will generate the gcode which includes the data as comments:
```
...
; filament used [mm] = 14664.38
; filament used [cm3] = 35.27
; total filament used [g] = 0.00
; total filament cost = 0.00
; estimated printing time (normal mode) = 2h 36m 25s
...
```

I'm not sure if the 3mf format can include this information somehow.  
This would be probably the simplest solution.  
If this isn't possible maybe adding an additional CLI parameter like ``--export-estimations filepath.json`` can also be an option.

**Describe alternatives you've considered**
I tried to export the sliced file as 3mf file, replaced the 3mf extension with "zip" and searched for the data but couldn't find it.  
I've searched for official profiles which can be imported to PrucaSlicer (but actually I want to use BambuLab Studio for this).

**Additional context**
Estimation data in the preview tab of the GUI:
[image]

theogott, 2022-12-17:
If its commercial, you can use the Smart Package Robot to automate such stuff.
He can directly read all Data from the GUI. You can get it here.
[Smart Package Robot](http://www.smart-package.com/download.html)
PS: Its not Freeware but commercial.
[image]

sguter90, 2022-12-17:
@theogott Thanks for your suggestion, but I mainly use MacOS and Linux.   
Even if the Smart Package Robot would work on e.g. Linux this would need a desktop environment (which I don't have on most of my Linux machines).  
Setting up an additional Windows VM or something like that is an overkill in my opinion.  
This is the reason why this will not work for me.

theogott, 2022-12-17:
I understand, and agree that an open Interface from Bambu Studio will be an advantage as it will enable also a lot more Integrations.

FOREVEREALIZE, 2024-07-31:
So I found kind of a workaround:

`.3mf` files are just ZIP files in disguise. One can unzip them and, in the files within, there's a folder called `Metadata`. Inside this folder, the `slice_info.config` file, which is, once again, an XML file in disguise, contains the `filament` tag which, as one of its atributes, has `used_m` (millimeters used) and `used_g`.

If you know the price per kilogram, you can calculate the cost as: `price_kg / 1000 * used_g`.

Hope this helps!

N4r35h, 2024-11-17:
still not able to utilize the cli to slice a given stl/3mf file and gets its print time and filament estimate anyone has found a solution ?

@FOREVEREALIZE my 3mf files Metadata/slice_info.config doesnt have those fields

FOREVEREALIZE, 2024-11-17:
> still not able to utilize the cli to slice a given stl/3mf file and gets its print time and filament estimate anyone has found a solution ?
> 
> @FOREVEREALIZE my 3mf files Metadata/slice_info.config doesnt have those fields

Are you using the sliced 3mf or just “geometry” 3mf?

N4r35h, 2024-11-18:
> > still not able to utilize the cli to slice a given stl/3mf file and gets its print time and filament estimate anyone has found a solution ?
> > @FOREVEREALIZE my 3mf files Metadata/slice_info.config doesnt have those fields
> 
> Are you using the sliced 3mf (exported after setting config and slicing) or just “geometry” 3mf (as in downloaded from Maker World)?

no its not downloaded from maker world im slicing a model that i created that i have as stl in my system using this cli command
the .json files are the example files from https://github.com/bambulab/BambuStudio/wiki/Command-Line-Usage

`./Bambu_Studio_linux_fedora-v01.09.07.52.AppImage --arrange 3 --load-settings "machine.json;process.json" --load-filaments "filament.json" --debug 2 --export-3mf output_test.3mf ./<redacted>.stl --orient 1`

ran in my ubuntu 20.04 machine

output
```
[2024-11-18 07:17:24.391703] [0x00007177ea88ba40] [trace]   Initializing StaticPrintConfigs
[2024-11-18 07:17:24.512387] [0x00007177ea88ba40] [warning] cli mode, Current BambuStudio Version 01.09.07.52
[2024-11-18 07:17:24.533002] [0x00007177ea88ba40] [warning] no filament colors found in projects
                                      overhang, bottom, bothull, contour, A_laf, A_prj, unprintability
orientation:-0.0000 -0.0000  1.0000, cost:234.9,	5878.5,	3967.2,	306.7,	27.5,	0.0,	0.1
orientation:-0.0000 -0.0000 -1.0000, cost:4148.6,	20.5,	188.5,	18.1,	93.1,	0.0,	140.3
orientation:-0.0000  1.0000 -0.0000, cost:919.8,	864.0,	594.1,	117.6,	74.8,	0.0,	1.4
orientation: 0.0000 -1.0000 -0.0000, cost:919.0,	868.9,	594.1,	117.9,	74.8,	0.0,	1.3
orientation:-1.0000 -0.0000 -0.0000, cost:586.8,	516.0,	431.0,	90.9,	48.3,	0.0,	1.4
orientation: 1.0000 -0.0000 -0.0000, cost:597.6,	481.7,	433.0,	87.8,	48.3,	0.0,	1.5
orientation: 0.5000  0.8660 -0.0000, cost:1451.9,	8.2,	8.2,	11.4,	31.6,	0.0,	171.6
orientation:-0.5000 -0.8660 -0.0000, cost:2045.5,	8.2,	8.2,	11.5,	31.6,	0.0,	240.8
orientation: 0.5000 -0.8660 -0.0000, cost:1430.4,	8.2,	8.3,	11.5,	33.6,	0.0,	168.4
orientation:-0.5000  0.8660 -0.0000, cost:1447.9,	8.2,	8.2,	11.5,	33.6,	0.0,	170.2
orientation:-0.0000  0.0165 -0.9999, cost:4028.7,	212.3,	2312.3,	58.3,	79.7,	0.0,	13.1
orientation: 0.0000  0.0165 -0.9999, cost:4028.7,	212.3,	2312.3,	58.3,	79.7,	0.0,	13.1
orientation:-0.0000 -0.0234 -0.9997, cost:4022.3,	204.9,	1687.1,	57.3,	81.7,	0.0,	15.3
orientation: 0.0000 -0.0234 -0.9997, cost:4022.3,	204.9,	1687.1,	57.3,	81.7,	0.0,	15.3
orientation: 0.0924 -0.0000 -0.9957, cost:4152.1,	15.9,	222.9,	15.9,	7666.1,	0.0,	159.0
orientation:-0.0985 -0.0000 -0.9951, cost:4153.7,	14.9,	211.6,	15.5,	7664.1,	0.0,	167.9
orientation:0.0000 0.1305 0.9914, cost:4093.5,	66.9,	57.4,	32.7,	7655.0,0.0,	77.0
orientation:-0.0000 -0.1305  0.9914, cost:4093.5,	67.0,	57.5,	32.7,	7655.0,	0.0,	76.9
orientation:-0.7071 -0.0000  0.7071, cost:4727.4,	55.8,	55.9,	29.9,	16.3,	0.0,	97.1
orientation:-0.0000 -0.7071  0.7071, cost:5295.0,	76.6,	76.7,	35.0,	36.9,	0.0,	80.4
orientation: 0.7071 -0.0000  0.7071, cost:4725.7,	55.7,	55.7,	29.9,	16.5,	0.0,	97.2
orientation:-0.0000  0.7071  0.7071, cost:5294.1,	76.7,	76.6,	35.0,	36.5,	0.0,	80.4
orientation:-0.7071 -0.7071 -0.0000, cost:2024.8,	8.2,	8.1,	11.5,	32.6,	0.0,	238.3
orientation: 0.7071 -0.7071 -0.0000, cost:2023.9,	8.2,	8.1,	11.5,	33.3,	0.0,	238.2
orientation: 0.7071  0.7071 -0.0000, cost:2023.9,	8.2,	8.3,	11.5,	32.6,	0.0,	237.6
orientation:-0.7071  0.7071 -0.0000, cost:2026.4,	8.2,	8.1,	11.4,	33.3,	0.0,	239.2
orientation:-0.7071 -0.0000 -0.7071, cost:4735.5,	0.0,	0.0,	0.0,	44.2,	0.0,	12219.6
orientation:-0.0000 -0.7071 -0.7071, cost:5302.7,	14.7,	0.0,	15.4,	74.5,	0.0,	398.7
orientation: 0.7071 -0.0000 -0.7071, cost:4736.1,	0.0,	0.0,	0.0,	44.0,	0.0,	12221.2
orientation:-0.0000  0.7071 -0.7071, cost:5312.1,	0.0,	0.0,	0.7,	75.0,	0.0,	10643.6
best:-0.000000 -0.000000  1.000000, costs:234.9,	5878.5,	3967.2,	306.7,	27.5,	0.0,	0.1
[2024-11-18 07:17:24.552325] [0x00007177ea88ba40] [error]   error_code 65544, description: Wayland: Failed to connect to display

[2024-11-18 07:17:24.552337] [0x00007177ea88ba40] [error]   glfwInit return error, code 65544
```

the output_test.3mf/Metadata/slice_info.config has this content inside it
```
<?xml version="1.0" encoding="UTF-8"?>
<config>
  <header>
    <header_item key="X-BBL-Client-Type" value="slicer"/>
    <header_item key="X-BBL-Client-Version" value="01.09.07.52"/>
  </header>
</config>
```

N4r35h, 2024-11-18:
> That's odd, I've just tried it with my BambuStudio and it works. I am
> running a bit of an older version (maybe one or two releases) so that might
> be it. Try a couple older versions (assuming the one you are using is
> latest. If it's not, try the last couple releases).

Thanks, will give it a shot today once I get some time and update here, if
possible can you show an example of how that file would be ?

FOREVEREALIZE, 2024-11-18:
> Thanks, will give it a shot today once I get some time and update here, if possible can you show an example of how that file would be ?

Sure! Here is the Bambu Filament Spool (not original one) sliced for the X1C 0.4mm nozzle with Generic PLA as the filament.

(It's in a zip file because GitHub doesn't like `.config` files...)

[slice_info.config.zip](https://github.com/user-attachments/files/17796929/slice_info.config.zip)

N4r35h, 2024-11-18:
Thanks, also how should I go about getting json files for my printer which
is the a1 and it's standard configs ? is there a way or place to download
those configurations ? filament, process etc

N4r35h, 2024-11-18:
> That's odd, I've just tried it with my BambuStudio and it works. I am running a bit of an older version (maybe one or two releases) so that might be it. Try a couple older versions (assuming the one [...]

still not able to get it working tried latest and a month old release both doesnt work my slice_info.config is similar to the one i sent above.... can you share your preticular binary version ? im running ubunu 20.04 if that matters